espressif/esp32-arduino-lib-builder #1: libesp32.a does not update for APPTRACE

TB-DaveMoore, 2019-09-17:
### **Hardware**
Board: ESP32-WROOM
Core Installation version:
IDE name: VSCode/PlatformIO
Flash Frequency: 80 MHZ
PSRAM enabled: no
Upload Speed: 15200
Computer OS: Mac OSX

### Description
I have run the builder script to try and enable app trace with the following config settings

`CONFIG_ESP32_APPTRACE_DEST_TRAX=y`
`CONFIG_ESP32_APPTRACE_ENABLE=y`
`CONFIG_ESP32_APPTRACE_LOCK_ENABLE=y`
`CONFIG_ESP32_APPTRACE_ONPANIC_HOST_FLUSH_TMO=-1`
`CONFIG_ESP32_APPTRACE_POSTMORTEM_FLUSH_THRESH=0`
`CONFIG_ESP32_APPTRACE_PENDING_DATA_SIZE_MAX=0`

However, copying the resulting libesp32.a to PlatformIO's arduinoespressif32 framework still results in an undefined reference error when referencing `esp_apptrace_vprintf()`. I have also tried copying _all_ generated libraries, but that just results in further error. Looking through the readable parts of libesp32.a, I cannot see any reference to apptrace being enabled. 

### Code
```
#include <Arduino.h>
#include "esp_app_trace.h"
#include "esp_log.h"

void setup() {
  // put your setup code here, to run once:
  esp_log_set_vprintf(esp_apptrace_vprintf);
}

void loop() {
  // put your main code here, to run repeatedly:
  ESP_LOGV("TEST","Test log message");
  esp_apptrace_flush(ESP_APPTRACE_DEST_TRAX, 100000);
  vTaskDelay(1000/portTICK_PERIOD_MS);
}
```
### Compiler Messages
```
Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32dev; framework: arduino)
----------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.9.0 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (jlink) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), tool-openocd-esp32 1.1000.20190708 (10.0), framework-arduinoespressif32 2.10002.190628 (1.0.2), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 26 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pio/build/esp32doit-devkit-v1/src/main.cpp.o
Generating partitions .pio/build/esp32doit-devkit-v1/partitions.bin
Archiving .pio/build/esp32doit-devkit-v1/libFrameworkArduinoVariant.a
Indexing .pio/build/esp32doit-devkit-v1/libFrameworkArduinoVariant.a
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/Esp.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/FunctionalInterrupt.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/IPv6Address.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/MD5Builder.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/Print.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/Stream.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/StreamString.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/WMath.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/WString.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/base64.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/cbuf.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/esp32-hal-adc.c.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/esp32-hal-bt.c.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/esp32-hal-cpu.c.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/esp32-hal-dac.c.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/esp32-hal-gpio.c.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/esp32-hal-i2c.c.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/esp32-hal-ledc.c.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/esp32-hal-matrix.c.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/esp32-hal-misc.c.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/esp32-hal-psram.c.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/esp32-hal-rmt.c.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/esp32-hal-sigmadelta.c.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/esp32-hal-spi.c.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/esp32-hal-time.c.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/esp32-hal-timer.c.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/esp32-hal-touch.c.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/esp32-hal-uart.c.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/libb64/cdecode.c.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/libb64/cencode.c.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/main.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/stdlib_noniso.c.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/esp32doit-devkit-v1/libFrameworkArduino.a
Indexing .pio/build/esp32doit-devkit-v1/libFrameworkArduino.a
Linking .pio/build/esp32doit-devkit-v1/firmware.elf
.pio/build/esp32doit-devkit-v1/src/main.cpp.o:(.literal._Z5setupv+0x0): undefined reference to `esp_apptrace_vprintf(char const*, __va_list_tag)'
.pio/build/esp32doit-devkit-v1/src/main.cpp.o:(.literal._Z4loopv+0x4): undefined reference to `esp_apptrace_flush(esp_apptrace_dest_t, unsigned int)'
.pio/build/esp32doit-devkit-v1/src/main.cpp.o: In function `loop()':
main.cpp:(.text._Z4loopv+0x8): undefined reference to `esp_apptrace_flush(esp_apptrace_dest_t, unsigned int)'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32doit-devkit-v1/firmware.elf] Error 1
```

me-no-dev, 2019-09-18:
Maybe it's in another lib? Are you sure it ends up in libesp32? Why not copy the whole rebuild SDK by using `./tools/copy-to-arduino.sh` (adjust core path to your pio install). In any case, this has nothing to do with this repository :) it only builds and packs. Can not interfere in any way with IDF or what it builds

closed 2019-09-18 by me-no-dev

electrohighlow, 2020-10-22:
can anyone help me? Error 1

Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2.0.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.201016 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Arduino_JSON> 0.1.0
|-- <HTTPClient> 1.2
|   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <WiFi> 1.0
Building in release mode
Compiling .pio\build\esp32dev\src\main.cpp.o
src\main.cpp: In function 'void loop()':
src\main.cpp:49:47: error: 'httpGETRequest' was not declared in this scope
       outputsState = httpGETRequest(serverName);
                                               ^
Archiving .pio\build\esp32dev\lib8af\libHTTPClient.a
Compiling .pio\build\esp32dev\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\IPv6Address.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\MD5Builder.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\Print.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\Stream.cpp.o
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1
Compiling .pio\build\esp32dev\FrameworkArduino\StreamString.cpp.o
================================================================= [FAILED] Took 4.70 seconds ===
The terminal process "C:\Users\96654\.platformio\penv\Scripts\platformio.exe 'run'" terminated w