espressif/esp32-arduino-lib-builder #8: Failing to building on Raspberry Pi due to missing ffi

this-username-is-taken, 2020-01-02:
I got the error below when running `./build.sh` on Raspberry Pi. I was able to get around it by running `sudo apt-get install libffi6 libffi-dev`. Further, building the cryptography lib requires root access, so will need to run `sudo ./build.sh`.

I was able to build at the end. It would be nice to update the README so other people don't hit the same issue.

>   Downloading https://files.pythonhosted.org/packages/be/60/da377e1bed002716fb2d5d1d1cab720f298cb33ecff7bf7adea72788e4e4/cryptography-2.8.tar.gz (504kB)
     |████████████████████████████████| 512kB 2.0MB/s 
  Installing build dependencies ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python /usr/local/lib/python2.7/dist-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-UqPw9S/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- 'setuptools>=40.6.0' wheel 'cffi>=1.8,!=1.11.3; platform_python_implementation != '"'"'PyPy'"'"''
       cwd: None
  Complete output (142 lines):
  DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple
  Collecting setuptools>=40.6.0
    Downloading https://www.piwheels.org/simple/setuptools/setuptools-43.0.0-py2.py3-none-any.whl (583kB)
  Collecting wheel
    Downloading https://www.piwheels.org/simple/wheel/wheel-0.33.6-py2.py3-none-any.whl
  Collecting cffi!=1.11.3,>=1.8
    Downloading https://files.pythonhosted.org/packages/2d/bf/960e5a422db3ac1a5e612cb35ca436c3fc985ed4b7ed13a1b4879006f450/cffi-1.13.2.tar.gz (460kB)
  Collecting pycparser
    Downloading https://www.piwheels.org/simple/pycparser/pycparser-2.19-py2.py3-none-any.whl (111kB)
  Building wheels for collected packages: cffi
    Building wheel for cffi (setup.py): started
    Building wheel for cffi (setup.py): finished with status 'error'
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-umc2wH/cffi/setup.py'"'"'; __file__='"'"'/tmp/pip-install-umc2wH/cffi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-t0RVnb --python-tag cp27
         cwd: /tmp/pip-install-umc2wH/cffi/
    Complete output (56 lines):
    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'libffi' found
    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'libffi' found
    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'libffi' found
    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'libffi' found
    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'libffi' found
    running bdist_wheel
    running build
    running build_py
    creating build
    creating build/lib.linux-armv7l-2.7
    creating build/lib.linux-armv7l-2.7/cffi
    copying cffi/recompiler.py -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/__init__.py -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/lock.py -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/error.py -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/pkgconfig.py -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/ffiplatform.py -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/commontypes.py -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/backend_ctypes.py -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/vengine_cpy.py -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/cffi_opcode.py -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/model.py -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/verifier.py -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/vengine_gen.py -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/api.py -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/setuptools_ext.py -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/cparser.py -> build/lib.linux-armv7l-2.7/cffi
[... 60 lines cut ...]
      copying cffi/cffi_opcode.py -> build/lib.linux-armv7l-2.7/cffi
      copying cffi/model.py -> build/lib.linux-armv7l-2.7/cffi
      copying cffi/verifier.py -> build/lib.linux-armv7l-2.7/cffi
      copying cffi/vengine_gen.py -> build/lib.linux-armv7l-2.7/cffi
      copying cffi/api.py -> build/lib.linux-armv7l-2.7/cffi
      copying cffi/setuptools_ext.py -> build/lib.linux-armv7l-2.7/cffi
      copying cffi/cparser.py -> build/lib.linux-armv7l-2.7/cffi
      copying cffi/_cffi_include.h -> build/lib.linux-armv7l-2.7/cffi
      copying cffi/parse_c_type.h -> build/lib.linux-armv7l-2.7/cffi
      copying cffi/_embedding.h -> build/lib.linux-armv7l-2.7/cffi
      copying cffi/_cffi_errors.h -> build/lib.linux-armv7l-2.7/cffi
      running build_ext
      building '_cffi_backend' extension
      creating build/temp.linux-armv7l-2.7
      creating build/temp.linux-armv7l-2.7/c
      arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-kKRR4y/python2.7-2.7.13=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DUSE__THREAD -DHAVE_SYNC_SYNCHRONIZE -I/usr/include/ffi -I/usr/include/libffi -I/usr/include/python2.7 -c c/_cffi_backend.c -o build/temp.linux-armv7l-2.7/c/_cffi_backend.o
      c/_cffi_backend.c:15:17: fatal error: ffi.h: No such file or directory
       #include <ffi.h>
                       ^
      compilation terminated.
      error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
      ----------------------------------------
  ERROR: Command errored out with exit status 1: /usr/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-umc2wH/cffi/setup.py'"'"'; __file__='"'"'/tmp/pip-install-umc2wH/cffi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-ENZgtD/install-record.txt --single-version-externally-managed --prefix /tmp/pip-build-env-UqPw9S/overlay --compile Check the logs for full command output.
  ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python /usr/local/lib/python2.7/dist-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-UqPw9S/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- 'setuptools>=40.6.0' wheel 'cffi>=1.8,!=1.11.3; platform_python_implementation != '"'"'PyPy'"'"'' Check the logs for full command output.

me-no-dev, 2020-01-02:
this looks related to instlling the IDF tools and not this repository :) maybe do not use python 2.7? I had no such issues on my PI

this-username-is-taken, 2020-01-03:
That could totally be the case. I will close this issue as the solution has been documented above.

closed 2020-01-03 by this-username-is-taken