espressif/esp32-arduino-lib-builder #10: Compilation error

GianCann, 2020-02-22:
I try to compiling esp32-arduino library following this guide:
https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md

I get this error:

/home/gianluca/esp32-arduino-lib-builder/components/arduino/libraries/WiFi/src/WiFiSTA.cpp: In member function 'IPAddress WiFiSTAClass::dnsIP(uint8_t)':
/home/gianluca/esp32-arduino-lib-builder/components/arduino/libraries/WiFi/src/WiFiSTA.cpp:491:37:  **error: conversion from 'const ip_addr_t* {aka const ip_addr*}' to non-scalar type 'ip_addr_t {aka ip_addr}' requested **
     ip_addr_t dns_ip = dns_getserver(dns_no);
                                     ^
/home/gianluca/esp32-arduino-lib-builder/components/arduino/libraries/WiFi/src/ETH.cpp: In member function 'IPAddress ETHClass::dnsIP(uint8_t)':
/home/gianluca/esp32-arduino-lib-builder/components/arduino/libraries/WiFi/src/ETH.cpp:196:37:  **error: conversion from 'const ip_addr_t* {aka const ip_addr*}' to non-scalar type 'ip_addr_t {aka ip_addr}' requested**
     ip_addr_t dns_ip = dns_getserver(dns_no);

Where is the problem?

fc92, 2020-04-10:
@ivanporro proposed a [workaround](https://github.com/espressif/arduino-esp32/issues/3760) that worked for me

me-no-dev, 2020-04-10:
this has been fixed recently. it was because the api changed.

me-no-dev, 2020-04-10:
well... not merged because of other reasons. but will be quite soon.

closed 2020-04-10 by me-no-dev