espressif/esp32-arduino-lib-builder #12: 3 Compilation Errors

fededim, 2020-05-05:
I am trying to build the Arduino library with RTC source changed from 150Khz RC Oscillator to external crystal (all the other settings are default). Os is Ubuntu 18.04.

Here follows my compilation errors:

**1)** 

```
/home/compile/esp32-arduino-lib-builder/components/esp32-camera/driver/camera.c: In function 'esp_camera_save_to_nvs':
/home/compile/esp32-arduino-lib-builder/components/esp32-camera/driver/camera.c:1429:5: error: unknown type name 'nvs_handle_t'
     nvs_handle_t handle;
     ^
/home/compile/esp32-arduino-lib-builder/components/esp32-camera/driver/camera.c:1433:48: error: pointer targets in passing argument 3 of 'nvs_open' differ in signedness [-Werror=pointer-sign]
     esp_err_t ret = nvs_open(key,NVS_READWRITE,&handle);
                                                ^
In file included from /home/compile/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/include/nvs_flash.h:21:0,
                 from /home/compile/esp32-arduino-lib-builder/components/esp32-camera/driver/camera.c:32:
/home/compile/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/include/nvs.h:109:11: note: expected 'nvs_handle * {aka unsigned int *}' but argument is of type 'int *'
 esp_err_t nvs_open(const char* name, nvs_open_mode open_mode, nvs_handle *out_handle);
           ^
/home/compile/esp32-arduino-lib-builder/components/esp32-camera/driver/camera.c: In function 'esp_camera_load_from_nvs':
/home/compile/esp32-arduino-lib-builder/components/esp32-camera/driver/camera.c:1457:5: error: unknown type name 'nvs_handle_t'
     nvs_handle_t handle;
     ^
/home/compile/esp32-arduino-lib-builder/components/esp32-camera/driver/camera.c:1463:46: error: pointer targets in passing argument 3 of 'nvs_open' differ in signedness [-Werror=pointer-sign]
   esp_err_t ret = nvs_open(key,NVS_READWRITE,&handle);
                                              ^
In file included from /home/compile/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/include/nvs_flash.h:21:0,
                 from /home/compile/esp32-arduino-lib-builder/components/esp32-camera/driver/camera.c:32:
/home/compile/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/include/nvs.h:109:11: note: expected 'nvs_handle * {aka unsigned int *}' but argument is of type 'int *'
 esp_err_t nvs_open(const char* name, nvs_open_mode open_mode, nvs_handle *out_handle);
           ^
cc1: some warnings being treated as errors
/home/compile/esp32-arduino-lib-builder/esp-idf/make/component_wrapper.mk:289: recipe for target 'driver/camera.o' failed
make[1]: *** [driver/camera.o] Error 1
/home/compile/esp32-arduino-lib-builder/esp-idf/make/project.mk:552: recipe for target 'component-esp32-camera-build' failed
```

**2)**

```
/home/compile/esp32-arduino-lib-builder/components/arduino/libraries/WiFi/src/ETH.cpp: In member function 'IPAddress ETHClass::dnsIP(uint8_t)':
/home/compile/esp32-arduino-lib-builder/components/arduino/libraries/WiFi/src/ETH.cpp:196:37: error: conversion from 'const ip_addr_t* {aka const ip_addr*}' to non-scalar type 'ip_addr_t {aka ip_addr}' requested
     ip_addr_t dns_ip = dns_getserver(dns_no);
                                     ^
/home/compile/esp32-arduino-lib-builder/esp-idf/make/component_wrapper.mk:289: recipe for target 'libraries/WiFi/src//ETH.o' failed
make[1]: *** [libraries/WiFi/src//ETH.o] Error 1

```

**3)**

```
/home/compile/esp32-arduino-lib-builder/components/arduino/libraries/WiFi/src/WiFiSTA.cpp: In member function 'IPAddress WiFiSTAClass::dnsIP(uint8_t)':
/home/compile/esp32-arduino-lib-builder/components/arduino/libraries/WiFi/src/WiFiSTA.cpp:491:37: error: conversion from 'const ip_addr_t* {aka const ip_addr*}' to non-scalar type 'ip_addr_t {aka ip_addr}' requested
     ip_addr_t dns_ip = dns_getserver(dns_no);
                                     ^

```

me-no-dev, 2020-05-05:
that is fine. there are changes in IDF that broke the compilation. you can try to usethe following branch: https://github.com/espressif/arduino-esp32/tree/latest-3.3

fededim, 2020-05-13:
I managed to fix the errors and compiled everything. For those interested, here follows the patches:

WiFiSta.cpp
```
    const ip_addr_t* dns_ip = dns_getserver(dns_no);
    return IPAddress(dns_ip->u_addr.ip4.addr);
```

Eth.cpp
```
    const ip_addr_t* dns_ip = dns_getserver(dns_no);
    return IPAddress(dns_ip->u_addr.ip4.addr);
```

Camera.c

Row 1428
```
#ifdef ESP_IDF_VERSION_MAJOR
    nvs_handle handle;
#else
    nvs_handle handle;
#endif
```

Row 1456
```
#ifdef ESP_IDF_VERSION_MAJOR
    nvs_handle handle;
#else
    nvs_handle handle;
#endif

```

misghna, 2020-05-16:
@me-no-dev I have tried this (https://github.com/espressif/arduino-esp32/tree/latest-3.3), but a previously working code now throws the following error   

....ino.elf section `.iram0.text' will not fit in region `iram0_0_seg'
./xtensa-esp32-elf/bin/ld.exe: region `iram0_0_seg' overflowed by 7817 bytes
/bin/ld.exe: region `iram0_0_seg' overflowed by 7817 bytes

any help would be appreciated thanks :)

hassance3, 2020-05-22:
@fededim Thank you for your patches.

hassance3, 2020-05-22:
@misghna Please use fededim patches for now.

JLIspace, 2023-12-17:
I had a similar issue, using "nvs_handle" instead of "nvs_handle_t" fixed it!