espressif/esp32-arduino-lib-builder #17: can't build environement

lagunax, 2020-08-25:
jast trying to reinstall. removed all. next oing like it escribe here an got:./build.sh
```
.....many lines CC etc....

CXX build/arduino/libraries/WiFi/src//WiFiScan.o
/PATHTO/esp/esp32-arduino-lib-builder/components/arduino/libraries/WiFi/src/ETH.cpp: In member function 'IPAddress ETHClass::dnsIP(uint8_t)':
/PATHTO/esp/esp32-arduino-lib-builder/components/arduino/libraries/WiFi/src/ETH.cpp:196:37: error: conversion from 'const ip_addr_t* {aka const ip_addr*}' to non-scalar type 'ip_addr_t {aka ip_addr}' requested
     ip_addr_t dns_ip = dns_getserver(dns_no);
                                     ^
...many lines...
/PATHTO/esp/esp32-arduino-lib-builder/esp-idf/components/app_update/Makefile.projbuild:19: предупреждение: неопределённая переменная «quote»
make: *** [/PATHTO/esp/esp32-arduino-lib-builder/esp-idf/make/project.mk:552: component-arduino-build] Ошибка 2
Toolchain path: /PATHTO/esp/esp32-arduino-lib-builder/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-80-g6c4433a
Expected to see version: crosstool-ng-1.22.0-96-g2852398
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Compiler version: 5.2.0
Arduino libraries in build: ArduinoOTA AsyncUDP AzureIoT BLE BluetoothSerial DNSServer EEPROM ESP32 ESPmDNS FFat FS HTTPClient HTTPUpdate NetBIOS Preferences README.md SD SD_MMC SPI SPIFFS SimpleBLE Ticker Update WebServer WiFi WiFiClientSecure Wire
App "esp32-arduino-lib-builder" version: dca3e75-dirty
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-80-g6c4433a
Expected to see version: crosstool-ng-1.22.0-96-g2852398
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Python requirements from /PATHTO/esp/esp32-arduino-lib-builder/esp-idf/requirements.txt are satisfied.

...some more lines...

/PATHTO/esp/esp32-arduino-lib-builder/esp-idf/components/app_update/Makefile.projbuild:19: предупреждение: неопределённая переменная «quote»
/PATHTO/esp/esp32-arduino-lib-builder/components/arduino/libraries/WiFi/src/WiFiSTA.cpp: In member function 'IPAddress WiFiSTAClass::dnsIP(uint8_t)':
/PATHTO/esp/esp32-arduino-lib-builder/components/arduino/libraries/WiFi/src/WiFiSTA.cpp:491:37: error: conversion from 'const ip_addr_t* {aka const ip_addr*}' to non-scalar type 'ip_addr_t {aka ip_addr}' requested
     ip_addr_t dns_ip = dns_getserver(dns_no);
                                     ^
make[1]: *** [/PATHTO/esp/esp32-arduino-lib-builder/esp-idf/make/component_wrapper.mk:290: libraries/WiFi/src//WiFiSTA.o] Ошибка 1
make[1]: *** Ожидание завершения заданий…
/PATHTO/esp/esp32-arduino-lib-builder/components/arduino/libraries/WiFi/src/ETH.cpp: In member function 'IPAddress ETHClass::dnsIP(uint8_t)':
/PATHTO/esp/esp32-arduino-lib-builder/components/arduino/libraries/WiFi/src/ETH.cpp:196:37: error: conversion from 'const ip_addr_t* {aka const ip_addr*}' to non-scalar type 'ip_addr_t {aka ip_addr}' requested
     ip_addr_t dns_ip = dns_getserver(dns_no);
                                     ^
make[1]: *** [/PATHTO/esp/esp32-arduino-lib-builder/esp-idf/make/component_wrapper.mk:290: libraries/WiFi/src//ETH.o] Ошибка 1
make: *** [/PATHTO/esp/esp32-arduino-lib-builder/esp-idf/make/project.mk:552: component-arduino-build] Ошибка 2

```

does i nee to clone older buil?

lagunax, 2020-08-25:
```
git checkout 97d7d20git checkout 97d7d20
./build.sh 

...manylines...

/PATHTO/esp/esp32-arduino-lib-builder/components/arduino/libraries/WiFi/src/ETH.cpp: In member function 'IPAddress ETHClass::dnsIP(uint8_t)':
/PATHTO/esp/esp32-arduino-lib-builder/components/arduino/libraries/WiFi/src/ETH.cpp:196:37: error: conversion from 'const ip_addr_t* {aka const ip_addr*}' to non-scalar type 'ip_addr_t {aka ip_addr}' requested
     ip_addr_t dns_ip = dns_getserver(dns_no);
                                     ^

/PATHTO/esp/esp32-arduino-lib-builder/esp-idf/components/app_update/Makefile.projbuild:19: предупреждение: неопределённая переменная «quote»
make[1]: *** [/PATHTO/esp/esp32-arduino-lib-builder/esp-idf/make/component_wrapper.mk:289: libraries/WiFi/src//ETH.o] Ошибка 1
make: *** [/PATHTO/esp/esp32-arduino-lib-builder/esp-idf/make/project.mk:552: component-arduino-build] Ошибка 2
```

lbernstone, 2020-09-05:
dupe
https://github.com/espressif/esp32-arduino-lib-builder/issues/12