espressif/esp32-arduino-lib-builder #19: Build successfull but image is not working

alorbach, 2020-12-01:
I finally got a the environment build successfully and working within Arduino CC  by changing the arduino-esp32 branch to "idf-release/v3.3" (See https://github.com/espressif/esp32-arduino-lib-builder/issues/12 as well)

I made these changes in my sdkconfig_: 
```
#
# Bluetooth controller
#

CONFIG_BTDM_CONTROLLER_MODE_BLE_ONLY=y

#
# Wi-Fi
#

CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=64
CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM=64
CONFIG_ESP32_WIFI_TX_BA_WIN=32
CONFIG_ESP32_WIFI_RX_BA_WIN=32

#
# TLS Key Exchange Methods
#

CONFIG_MBEDTLS_SSL_PROTO_DTLS=y

```

Finally after I build an image, the build is not working as expected :(. 
DHCP is not working Serial Input also does not work. 

Is there any chance to make a working build of the libs with these changed sdkconfig defined?

alorbach, 2020-12-02:
Well for now I switched back to old esp-idf 3.2 and arduino-esp32 tag 1.0.4 with the bugfix from 
https://github.com/espressif/esp32-arduino-lib-builder/issues/12

Now I can finally build my own arduino sdk with changed options. 
https://github.com/alorbach/esp32-arduino-lib-builder/commit/2c682aaf63f4dbdb9469a6d462dd39289492d93f

I even tried esp-idf 3.3 with arduino-esp32 tag 1.0.5rc3 from yesterday, but It the build comes out with the same problem as in my initial ticket, it's not getting an IP address from DHCP.

me-no-dev, 2020-12-02:
ok, lets clarify :) build is OK, but WiFi is not? If you are using the master branch of this repo, `./build.sh && ESP32_ARDUINO="PATH_TO_ESP32_ARDUINO" ./tools/copy-to-arduino.sh` would update your Arduino install just fine. This is exactly how I updated to RC3 yesterday

alorbach, 2020-12-02:
Yes I could build successfull with esp-idf branch release/v3.3 and arduino-esp32 tag 1.0.5rc3. 
However DHCP is not working for some reason. 

I am on the old working stable for now.

The reason why I wanted to check for newer esp-idf /arduino-esp32 is that I have strange WLAN disconnect problems which random are happening under UDP load. 

I am sending about 50 UDP packets (about 40 bytes each) per second by DTLS Streaming (MBEDTLS needs ) and I am receiving about 60 UDP packets (about 500 bytes each) RAW using AsyncUDP. 

At some random point the DTLS stream becomes unstable and shortly after this WIFI becomes unreliable until reboot, for example here is some debug output with automatic Wifi.Reconnect() tries between. 

> > %debug: 0:4126256: esp32debug: I (4126462) wifi:%debug: 0:4126256: esp32debug: state: assoc -> init (400)%debug: 0:4126256: esp32debug: 
> %debug: 0:4126257: esp32debug: I (4126464) wifi:%debug: 0:4126261: esp32debug: new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
> %debug: 1:4126272: wpsWiFiEvent:Disconnected: Disconnected from station reason: 4 (assoc expire), call esp_wifi_connect 
> %debug: 1:4126282: wpsWiFiEvent:Disconnected: WiFi.reconnect
> %debug: 1:4128338: wpsWiFiEvent:Disconnected: Disconnected from station reason: 205 (other reason), call esp_wifi_connect 
> %debug: 1:4128338: wpsWiFiEvent:Disconnected: WiFi.reconnect
> %debug: 0:4128704: esp32debug: I (4128911) wifi:%debug: 0:4128704: esp32debug: new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
> %debug: 0:4128708: esp32debug: I (4128915) wifi:%debug: 0:4128712: esp32debug: state: init -> auth (b0)%debug: 0:4128717: esp32debug: 
> %debug: 0:4129720: esp32debug: I (4129927) wifi:%debug: 0:4129720: esp32debug: state: auth -> init (200)%debug: 0:4129721: esp32debug: 
> %debug: 0:4129721: esp32debug: I (4129928) wifi:%debug: 0:4129725: esp32debug: new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
> %debug: 1:4129737: wpsWiFiEvent:Disconnected: Disconnected from station reason: 2 (auth expire), call esp_wifi_connect 
> %debug: 1:4129746: wpsWiFiEvent:Disconnected: WiFi.reconnect
> %debug: 0:4130112: esp32debug: I (4130319) wifi:%debug: 0:4130112: esp32debug: new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
> %debug: 0:4130116: esp32debug: I (4130323) wifi:%debug: 0:4130120: esp32debug: state: init -> auth (b0)%debug: 0:4130125: esp32debug: 
> %debug: 0:4131128: esp32debug: I (4131335) wifi:%debug: 0:4131128: esp32debug: state: auth -> init (200)%debug: 0:4131128: esp32debug: 
> %debug: 0:4131129: esp32debug: I (4131336) wifi:%debug: 0:4131133: esp32debug: new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
> %debug: 1:4131145: wpsWiFiEvent:Disconnected: Disconnected from station reason: 2 (auth expire), call esp_wifi_connect 
> %debug: 1:4131154: wpsWiFiEvent:Disconnected: WiFi.reconnect
> %debug: 0:4131520: esp32debug: I (4131726) wifi:%debug: 0:4131520: esp32debug: new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
> %debug: 0:4131524: esp32debug: I (4131730) wifi:%debug: 0:4131528: esp32debug: state: init -> auth (b0)%debug: 0:4131532: esp32debug: 
> %debug: 0:4132536: esp32debug: I (4132742) wifi:%debug: 0:4132536: esp32debug: state: auth -> init (200)%debug: 0:4132536: esp32debug: 
> %debug: 0:4132537: esp32debug: I (4132744) wifi:%debug: 0:4132541: esp32debug: new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
> %debug: 1:4132553: wpsWiFiEvent:Disconnected: Disconnected from station reason: 2 (auth expire), call esp_wifi_connect 
> %debug: 1:4132562: wpsWiFiEvent:Disconnected: WiFi.reconnect
> %debug: 0:4132928: esp32debug: I (4133134) wifi:%debug: 0:4132928: esp32debug: new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
> %debug: 0:4132932: esp32debug: I (4133138) wifi:%debug: 0:4132936: esp32debug: state: init -> auth (b0)%debug: 0:4132940: esp32debug: 
> %debug: 0:4133944: esp32debug: I (4134150) wifi:%debug: 0:4133944: esp32debug: state: auth -> init (200)%debug: 0:4133944: esp32debug: 
> %debug: 0:4133945: esp32debug: I (4134152) wifi:%debug: 0:4133949: esp32debug: new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
> %debug: 1:4133961: wpsWiFiEvent:Disconnected: Disconnected from station reason: 2 (auth expire), call esp_wifi_connect 
> %debug: 1:4133969: wpsWiFiEvent:Disconnected: WiFi.reconnect
> %debug: 0:4134336: esp32debug: I (4134542) wifi:%debug: 0:4134336: esp32debug: new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
> %debug: 0:4134340: esp32debug: I (4134546) wifi:%debug: 0:4134344: esp32debug: state: init -> auth (b0)%debug: 0:4134348: esp32debug: 
> %debug: 0:4134355: esp32debug: I (4134561) wifi:%debug: 0:4134355: esp32debug: state: auth -> assoc (0)%debug: 0:4134360: esp32debug: 
> %debug: 0:4135363: esp32debug: I (4135570) wifi:%debug: 0:4135364: esp32debug: state: assoc -> init (400)%debug: 0:4135364: esp32debug: 
> %debug: 0:4135365: esp32debug: I (4135571) wifi:%debug: 0:4135369: esp32debug: new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
> %debug: 1:4135380: wpsWiFiEvent:Disconnected: Disconnected from station reason: 4 (assoc expire), call esp_wifi_connect 
> %debug: 1:4135389: wpsWiFiEvent:Disconnected: WiFi.reconnect
> %debug: 0:4135834: esp32debug: W (4136040) wifi:%debug: 0:4135834: esp32debug: Haven't to connect to a suitable AP now!%debug: 0:4135834: esp32debug: 
> %debug: 1:4137446: wpsWiFiEvent:Disconnected: Disconnected from station reason: 205 (other reason), call esp_wifi_connect 
> %debug: 1:4137446: wpsWiFiEvent:Disconnected: WiFi.reconnect
> %debug: 0:4137812: esp32debug: I (4138018) wifi:%debug: 0:4137812: esp32debug: new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
> %debug: 0:4137816: esp32debug: I (4138022) wifi:%debug: 0:4137820: esp32debug: state: init -> auth (b0)%debug: 0:4137824: esp32debug: 
> %debug: 0:4137832: esp32debug: I (4138039) wifi:%debug: 0:4137832: esp32debug: state: auth -> assoc (0)%debug: 0:4137836: esp32debug:

me-no-dev, 2020-12-02:
have you checked how much memory you have available?

alorbach, 2020-12-02:
@me-no-dev: yes sure herr is some debug output from between the reconnect tries: 

> %debug: 1:4230853: loop: WiFi=Disconnected WifiSignal=-67dBm VDD=6678 Free/Heap/PSRAM=137KB/141KB/0KB LedMode=0 LedFPS=0/s Delay=30ms NetState=Waiting... NetPkts=0/0/s HueFPS=0/s  Tasks=16 

And when its own load it looks like this: 
```
%debug: 1:3810768: loop: WiFi=Connected WifiSignal=-69dBm VDD=6680 Free/Heap/PSRAM=100KB/103KB/0KB LedMode=1 LedFPS=55/s Delay=30ms NetState=Connected to 172.21.0.15 NetPkts=55/1/s HueFPS=26/s  Tasks=16 
%debug: 1:3815769: loop: WiFi=Connected WifiSignal=-71dBm VDD=6680 Free/Heap/PSRAM=100KB/103KB/0KB LedMode=1 LedFPS=55/s Delay=30ms NetState=Connected to 172.21.0.15 NetPkts=56/1/s HueFPS=26/s  Tasks=16 
%debug: 1:3820770: loop: WiFi=Connected WifiSignal=-70dBm VDD=6679 Free/Heap/PSRAM=100KB/103KB/0KB LedMode=1 LedFPS=55/s Delay=30ms NetState=Connected to 172.21.0.15 NetPkts=55/1/s HueFPS=26/s  Tasks=16 
```

alorbach, 2020-12-10:
For your information, I found the reeason. 
I had this line in my setup() since the beginning: 
`WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);`

It appears to overwrite the DHCP address in the new esp-idf.

closed 2020-12-10 by alorbach