espressif/esp32-arduino-lib-builder #20: "IRAM0 segment data does not fit"

alorbach, 2020-12-03:
Not sure if this is the right location for this issue. 
I am able to build based on esp-idf 3.2 and arduino-esp32 tag 1.0.4. 

I incremented some WIFI buffers to this: 
CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=32
CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM=32
CONFIG_ESP32_WIFI_TX_BA_WIN=12
CONFIG_ESP32_WIFI_RX_BA_WIN=16

Now when I build for a PSRAM enabled device (With Camera Support), I get this error: 
Controller.ino.elf section `.iram0.text' will not fit in region `iram0_0_seg'
xtensa-esp32-elf/bin/ld.exe: IRAM0 segment data does not fit.
xtensa-esp32-elf/bin/ld.exe: region `iram0_0_seg' overflowed by 2188 bytes

Is there anything I can to avoid this problem? 
CONFIG_ESP32_WIFI_IRAM_OPT and CONFIG_ESP32_WIFI_RX_IRAM_OPT are already disabled.

me-no-dev, 2020-12-03:
why not use 3.3 and master of this repo? Try with BT disabled (if it's enabled).

alorbach, 2020-12-03:
See https://github.com/espressif/esp32-arduino-lib-builder/issues/19
3.3 is not stable with my Sketch, I may try it again soon. 

I have workaround somehow for now by disabling CONFIG_SPIRAM_CACHE_WORKAROUND for now, see my settings: 
https://github.com/alorbach/esp32-arduino-lib-builder/commit/8f9ccd43af86e40b18b79cde012f61c3840574ed

It freed up enough IRAM memory so I can build by sketch again for esp32 cam modules.
However I would rather find another solution or perhaps increase double IRAM memory size if that makes sense?

I am using a lot of PSTR() to store debug printfs into flash memory. And my sketch has been grown alot, could this be related?

alorbach, 2020-12-16:
FYI: Fixed by building with esp-idf v3.3

closed 2020-12-16 by alorbach