espressif/esp32-arduino-lib-builder #21: Save to build with IDF_BRANCH release/v4.x?

alorbach, 2020-12-16:
I am on esp-idf 3.3 and arduino-esp32 master branch right now. 
Before II was on esp-idf 3.2. 

I am getting mad finding an "out of memory" inside LWIP which happens at a random point when using DTLS UDP Streaming (MBEDTLS). See my other issues:
https://github.com/espressif/esp-idf/issues/6241
https://github.com/ARMmbed/mbedtls/issues/3954

I have debugged so much into this and I am not able to find the reason why malloc() inside LWIP fails at a random point. 

My question is, is it save or possible to build against esp-idf 4.x or even the master branch, as there are already new heap api calls like heap_caps_register_failed_alloc_callback which I want to utlitize.

me-no-dev, 2020-12-16:
to build against master IDF, use the `release/v4.2` branch of this repo :)

alorbach, 2020-12-16:
thanks a lot :) I will try this and see if the memory problem is happening with newer esp-idf as well.

alorbach, 2020-12-18:
questions regarding v4.2 branch,:

- Is it python3 stable now?
- where has make menuconfig gone?

I am getting this when I run build.sh

> ESP-IDF is already installed at: ./esp-idf
> Note: You are using Python 3.6.9. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
> usage: idf.py [-h] [-p PORT] [-b BAUD] [-C PROJECT_DIR] [-B BUILD_DIR]
>               [-G {Ninja,Unix Makefiles}] [-n] [-v]
>               [-D DEFINE_CACHE_ENTRY [DEFINE_CACHE_ENTRY ...]] [--no-ccache]
>               {all,build,clean,fullclean,reconfigure,menuconfig,confserver,size,size-components,size-files,bootloader,bootloader-clean,bootloader-flash,app,app-flash,efuse_common_table,efuse_custom_table,show_efuse_table,partition_table,partition_table-flash,flash,erase_flash,monitor,erase_otadata,read_otadata}
>               [{all,build,clean,fullclean,reconfigure,menuconfig,confserver,size,size-components,size-files,bootloader,bootloader-clean,bootloader-flash,app,app-flash,efuse_common_table,efuse_custom_table,show_efuse_table,partition_table,partition_table-flash,flash,erase_flash,monitor,erase_otadata,read_otadata} ...]
> idf.py: error: argument actions: invalid choice: 'idf-libs' (choose from 'all', 'build', 'clean', 'fullclean', 'reconfigure', 'menuconfig', 'confserver', 'size', 'size-components', 'size-files', 'bootloader', 'bootloader-clean', 'bootloader-flash', 'app', 'app-flash', 'efuse_common_table', 'efuse_custom_table', 'show_efuse_table', 'partition_table', 'partition_table-flash', 'flash', 'erase_flash', 'monitor', 'erase_otadata', 'read_otadata')

me-no-dev, 2020-12-18:
Starting with IDF 4.0 the build system uses CMake and not Make. Proper config command is `idf.py menuconfig`. And yes, everything works with Python 3. Running `build.sh` should work if you have installed everything correctly.

alorbach, 2020-12-18:
Alright, I have that stuff installed from the readme but it puts out many errors like this:

> CMake Error at esp-idf/tools/cmake/idf_functions.cmake:29 (include):
>   include could not find load file:
> 
>     crosstool_version_check
> Call Stack (most recent call first):
>   esp-idf/tools/cmake/project.cmake:5 (include)
>   CMakeLists.txt:9 (include)
> 
> 
> CMake Error at esp-idf/tools/cmake/idf_functions.cmake:30 (include):
>   include could not find load file:
> 
>     ldgen
> Call Stack (most recent call first):
>   esp-idf/tools/cmake/project.cmake:5 (include)
>   CMakeLists.txt:9 (include)
> 
> 
> CMake Error at esp-idf/tools/cmake/idf_functions.cmake:31 (include):
>   include could not find load file:
> 
>     version
> Call Stack (most recent call first):
>   esp-idf/tools/cmake/project.cmake:5 (include)
>   CMakeLists.txt:9 (include)
> 
> 
> CMake Error at esp-idf/tools/cmake/idf_functions.cmake:33 (set_default):
>   Unknown CMake command "set_default".
> Call Stack (most recent call first):
>   esp-idf/tools/cmake/project.cmake:5 (include)
>   CMakeLists.txt:9 (include)

Is there new doc on the packages needed to be installed?

me-no-dev, 2020-12-18:
this is probably not a dependency issue.
To be clear:
- You cloned this repository
- Switched to release/v4.2 branch
- Executed `./build.sh`

alorbach, 2020-12-18:
Looks like I got the wrong xtensa-esp32-elf tools? 
I have downloaded them from here (linux-amd64) for v4.2: 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-tools.html

I unpacked it in my home dir in "xtensa-esp32-elf", and made sure it can be found by adding the path to the $PATH env: 
export PATH=${HOME}/xtensa-esp32-elf/bin:$PATH" 

I did the same for current master branch with esp-idf v3.3 

Now I am stuck here: 

> Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
> Compiler: /home/al/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc 
> Build flags: -mlongcalls
> Id flags:  
> 
> The output was:
> 1
> /home/al/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find crt1-sim.o: No such file or directory
> /home/al/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find _vectors.o: No such file or directory
> /home/al/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lsim
> /home/al/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lhandlers-sim
> /home/al/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lhal
> collect2: error: ld returned 1 exit status

...

alorbach, 2020-12-19:
@me-no-dev you were right, I still had wrong branch selected in esp-idf, seems like the build.sh script does not switch the branch automatically. 
I am a little step further now compiling, but now I am stuck here: 

> Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
> Compiler: xtensa-esp32s2-elf-gcc 
> Build flags: -mlongcalls
> Id flags:  
> 
> The output was:
> No such file or directory

hamishcunningham, 2020-12-19:
More recent versions of IDF install the s2 toolchain in ~/.espressif but
earlier ones don't. (I think it is esp-idf/install.sh that does this.)
Maybe run that script for the branch you're on, or even for a 4.3-dev?

hth
h

alorbach, 2020-12-22:
For now I got my code more or less stable with esp-idf 3.3 and arduino-esp32 master branch so I will properly try esp-idf v4.2 later when the building process is more stable or the build environment can be prepared automatically.

atecroby, 2021-08-06:
`release/v4.2` seems not present anymore
I need to build against v4.x, any idea on how to do that?

me-no-dev, 2021-08-09:
master branch builds against IDF master (4.4)

ecobdoor, 2022-08-05:
Hire
I know I'm not in the good place, but:
Is somebody able to give me examples to set the correct syntax to build.sh [-s] [-A arduino_branch] [-I idf_branch] [-i idf_commit] [-c path] [-t <target>] [-b <build|menuconfig|idf_libs|copy_bootloader|mem_variant>] [config ...]?
For example I want to build the arduino_branch version "2.0.1" based on  idf_branch "zzz":my only goal is to integrate vTaskGetRunTimeStats( char * pcWriteBuffer ). 

My project, already running, is a wifi rover, able of autonomous missions with RTK data & with 100 Hz IMU, using a javascript websocket browser interface (later an RF serial protocol...). All is correctly working in Eclipse & arduino IDE, environment, but I need vTaskGetRunTimeStats to optimize the tasks priorities/stacks/cores implementations (Lwip, esp_http_server & others...).

I try the custom configuration "./build.sh  -b menuconfig -t esp32" & it runs but I don't know how to set arduino & idf versions (option -A & -I).Can you help me?

Cordially yours (Eric retired computer design engineer https://github.com/ecobdoor/toto).