espressif/esp32-arduino-lib-builder #22: Question: git pull branch instead of checkout?

alorbach, 2020-12-18:
Is there any specific reason why the update scripts are using git pull instead of git checkout in the tools scripts?
When switching between branches, git pull causes a lot of unsolveable conflicts. 

https://github.com/espressif/esp32-arduino-lib-builder/blob/b86ba2bb50e2b1b0e55d7a530fffa228b5c2d841/tools/update-components.sh#L13

https://github.com/espressif/esp32-arduino-lib-builder/blob/master/tools/install-esp-idf.sh#L96

me-no-dev, 2020-12-18:
what is your use case? it uses pull so that the latest sources are pulled and updated, not so much for switching branches.

alorbach, 2020-12-18:
I think git pull is ok if you want to merge changes from one branch to another. 
Perhaps it should be changed 

`git -C "$AR_COMPS/arduino" checkout master && git pull `

`git -C $IDF_PATH fetch origin && git -C $IDF_PATH checkout $IDF_BRANCH && git pull origin`

For example when you switch from esp-idf 3.2 to 3.3 and back, you won't accidentally merge changes between both branches. 

FYI, I just noticed that the tools script also switched to git checkout in branch v4.2.

me-no-dev, 2020-12-18:
4.2 will become master very soon :) feel free to PR any changes you want to master now

alorbach, 2020-12-18:
I may create a PR later ok ;),
I will start looking into 4,2 now, if that's the new master soon :)

closed 2020-12-18 by alorbach