espressif/esp32-arduino-lib-builder #24: ESP32 Arduino 1.0.5 RC6 issue with lib-builder

RafigRzayev, 2021-01-14:
I have upgraded my Arduino IDE by adding this URL "https://github.com/espressif/arduino-esp32/releases/download/1.0.5-rc6/package_esp32_dev_index.json" to Additional Board Managers, so that Arduino version in lib-builder and the one in Arduino IDE match. After generating new libraries with the lib-builder tool, I replaced the "lib" folder in Arduino IDE with the one generated in lib-builder (inside "out" folder). Then I compiled my project and received following error: 
```
libraries\WiFi\WiFiGeneric.cpp.o:(.literal._ZN16WiFiGenericClass4modeE11wifi_mode_t+0x10): undefined reference to `g_wifi_feature_caps'
.
```

Then, instead of replacing just the "lib" folder, I replaced the whole "sdk" folder, and now everything compiles well. Is it normal behavior? Shouldn't updating just the lib folder be sufficient?

me-no-dev, 2021-01-14:
after building using `./build.sh` you should use `./tools/copy-to-arduino.sh` to copy everything to your Arduino install

closed 2021-01-14 by me-no-dev

RafigRzayev, 2021-01-14:
@me-no-dev its just I am using Windows. I have installed WSL for using Linux environment to generate the libraries, but I don't know how to use the copy-to-arduino.sh script to my Windows installation.