espressif/esp32-arduino-lib-builder #26: Error building release/v4.2

shaun-mzj, 2021-02-10:
I'm trying to compile the `release/v4.2` branch and am getting the following error: 

```
Executing action: idf-libs
Running cmake in directory /home/shaun/git/esp32-arduino-lib-builder/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32s2 -DCCACHE_ENABLE=0 /home/shaun/git/esp32-arduino-lib-builder"...
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/shaun/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/shaun/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/shaun/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s2
CMake Error at esp-idf/tools/cmake/build.cmake:186 (message):
  Failed to resolve component 'ethernet'.
Call Stack (most recent call first):
  esp-idf/tools/cmake/build.cmake:212 (__build_resolve_and_add_req)
  esp-idf/tools/cmake/build.cmake:426 (__build_expand_requirements)
  esp-idf/tools/cmake/project.cmake:395 (idf_build_process)
  CMakeLists.txt:10 (project)

-- Configuring incomplete, errors occurred!
See also "/home/shaun/git/esp32-arduino-lib-builder/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

All I've done is checkout this repo, switch branch and run the `./build/sh` script. I'm on: ubuntu 20.04

Is there anything else which needs to be done to build?

enwi, 2021-02-14:
I am having the same issue on MacOS Big Sur (11.2.1)

enwi, 2021-02-14:
I got it to compile for esp32 only. Steps:

1. Install esp-idf as explained [here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/) (steps 1 - 4)
2. Update esp-idf to `release/v4.3` following [this](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/versions.html#updating)
3. Clone `esp32-arduino-lib-builder` and make sure you are on branch `release/v4.2`
4. Modify `build.sh` and exchange `TARGETS="esp32s2 esp32"` with `TARGETS="esp32"`
5. Run `build.sh`

For step 1 it might be needed to enforce python3.

enwi, 2021-02-14:
But I am encountering even more errors using `esp32-arduino-lib-builder` in general on MacOS:

- awk commands are broken
```
Generating out
we are running in [/Users/moritz/esp/esp32-arduino-lib-builder] on IDF_TARGET: esp32, OUT: /Users/moritz/esp/esp32-arduino-lib-builder/out/tools/sdk/esp32
awk: syntax error at source line 1
 context is
	 >>> /ASFLAGS\=\[/{n++}{print>n"pio_start.txt" <<< 
awk: illegal statement at source line 1
awk: syntax error at source line 1
 context is
	 >>> /"ARDUINO_ARCH_ESP32"/{n++}{print>n"pio_end.txt" <<< 
awk: illegal statement at source line 1
cat: pio_start.txt: No such file or directory
rm: pio_end.txt: No such file or directory
rm: 1pio_start.txt: No such file or directory
rm: pio_start.txt: No such file or directory
cat: 1pio_end.txt: No such file or directory
rm: 1pio_end.txt: No such file or directory
awk: syntax error at source line 1
 context is
	 >>> /compiler.cpreprocessor.flags.esp32\=/{n++}{print>n"platform_start.txt" <<< 
awk: illegal statement at source line 1
gsed: can't read 1platform_start.txt: No such file or directory
awk: syntax error at source line 1
 context is
	 >>> /compiler.ar.flags.esp32\=/{n++}{print>n"platform_mid.txt" <<< 
awk: illegal statement at source line 1
cat: platform_start.txt: No such file or directory
cat: 1platform_mid.txt: No such file or directory
```
- even using gawk gives warnings:
```
Generating out
we are running in [/Users/moritz/esp/esp32-arduino-lib-builder] on IDF_TARGET: esp32, OUT: /Users/moritz/esp/esp32-arduino-lib-builder/out/tools/sdk/esp32
gawk: Kommandozeile:1: Warnung: Die Escapesequenz »\=« ist kein bekannter Operator für reguläre Ausdrücke
gawk: Kommandozeile:1: Warnung: Die Escapesequenz »\=« ist kein bekannter Operator für reguläre Ausdrücke
gawk: Kommandozeile:1: Warnung: Die Escapesequenz »\=« ist kein bekannter Operator für reguläre Ausdrücke
```
- The whole generated include folder structure is a mess and does not work with `arduino-esp32` giving numerous include errors
- The generated `platform.txt` misses a lot of definitions resulting in `recipe.preproc.macros pattern is missing` when trying to compile anything afterwards

maxgerhardt, 2021-02-20:
I'm getting the same 

>  Failed to resolve component 'ethernet'.

Error on Ubuntu 20.04. 

The solution to update ESP-IDF to v4.3 might work but my goal is to explicitly compile the v4.2 ESP-IDF libraries (with some sdkconfig modifications). 

This had to work at some point, otherwise https://github.com/espressif/arduino-esp32/tree/idf-release/v4.2/tools/sdk/esp32/lib couldn't have been generated by this last month. 

@lbernstone, could you have a look at this please?

Jason2866, 2021-03-01:
+1 for updating/fixing the 4.2 branch

Jason2866, 2021-04-03:
@me-no-dev thx for updating the 4.2 branch.
There is a issue left. For ESP32-S2 there is no ethernet.
```
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

-- Found Git: /usr/bin/git (found version "2.31.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/runner/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/runner/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/runner/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s2
CMake Warning at esp-idf/tools/cmake/build.cmake:472 (message):
  "idf_component.yml" file was found for components:

  
  	/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/esp32-camera

  However, the component manager is not enabled.
Call Stack (most recent call first):
  esp-idf/tools/cmake/project.cmake:358 (idf_build_process)
  CMakeLists.txt:10 (project)

CMake Error at esp-idf/tools/cmake/build.cmake:192 (message):
  Failed to resolve component 'ethernet'.
Call Stack (most recent call first):
  esp-idf/tools/cmake/build.cmake:218 (__build_resolve_and_add_req)
  esp-idf/tools/cmake/build.cmake:502 (__build_expand_requirements)
  esp-idf/tools/cmake/project.cmake:358 (idf_build_process)
  CMakeLists.txt:10 (project)

-- Configuring incomplete, errors occurred!
See also "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

me-no-dev, 2021-04-05:
this is temporary issue :) just remove `idf_component.yml` from the `esp32-camera` component

Jason2866, 2021-04-05:
Thx @me-no-dev , ESP32-camera warning is gone by deleting `idf_component.yml`. But the `ethernet` error is still there.
```
2021-04-05T10:44:37.3436701Z Done! You can now compile ESP-IDF projects.
2021-04-05T10:44:37.3437166Z Go to the project directory and run:
2021-04-05T10:44:37.3437418Z 
2021-04-05T10:44:37.3437681Z   idf.py build
2021-04-05T10:44:37.3437864Z 
2021-04-05T10:44:38.1644911Z -- Found Git: /usr/bin/git (found version "2.31.1") 
2021-04-05T10:44:40.3543009Z -- The C compiler identification is GNU 8.4.0
2021-04-05T10:44:40.4016257Z -- The CXX compiler identification is GNU 8.4.0
2021-04-05T10:44:40.4232891Z -- The ASM compiler identification is GNU
2021-04-05T10:44:40.4255358Z -- Found assembler: /home/runner/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc
2021-04-05T10:44:40.4777646Z -- Detecting C compiler ABI info
2021-04-05T10:44:40.5221648Z -- Detecting C compiler ABI info - done
2021-04-05T10:44:40.5299069Z -- Check for working C compiler: /home/runner/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc - skipped
2021-04-05T10:44:40.5301161Z -- Detecting C compile features
2021-04-05T10:44:40.5356426Z -- Detecting C compile features - done
2021-04-05T10:44:40.5453441Z -- Detecting CXX compiler ABI info
2021-04-05T10:44:40.5781948Z -- Detecting CXX compiler ABI info - done
2021-04-05T10:44:40.5867466Z -- Check for working CXX compiler: /home/runner/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++ - skipped
2021-04-05T10:44:40.5869646Z -- Detecting CXX compile features
2021-04-05T10:44:40.5875606Z -- Detecting CXX compile features - done
2021-04-05T10:44:40.5995313Z -- Building ESP-IDF components for target esp32s2
2021-04-05T10:44:41.1305820Z CMake Error at esp-idf/tools/cmake/build.cmake:192 (message):
2021-04-05T10:44:41.1306912Z   Failed to resolve component 'ethernet'.
2021-04-05T10:44:41.1307591Z Call Stack (most recent call first):
2021-04-05T10:44:41.1308723Z   esp-idf/tools/cmake/build.cmake:218 (__build_resolve_and_add_req)
2021-04-05T10:44:41.1309562Z   esp-idf/tools/cmake/build.cmake:502 (__build_expand_requirements)
2021-04-05T10:44:41.1310318Z   esp-idf/tools/cmake/project.cmake:358 (idf_build_process)
2021-04-05T10:44:41.1310850Z   CMakeLists.txt:10 (project)
2021-04-05T10:44:41.1311117Z 
2021-04-05T10:44:41.1311282Z 
2021-04-05T10:44:41.1314529Z -- Configuring incomplete, errors occurred!
```

me-no-dev, 2021-04-05:
are you sure you have everything correctly installed? I have no problem building and CI is working fine as well. BTW I just deleted the 4.2 branch (merged everything in master) so you might want to switch to that, pull arduino master as well (also updated) and try again

Jason2866, 2021-04-05:
Yes, because i do a clean sethub via github actions. It is a fork of your github. Just the branch changed in github actions to branch release/4.2 to trigger the build. See
https://github.com/Jason2866/esp32-arduino-lib-builder/tree/release/v4.2

me-no-dev, 2021-04-05:
try with current master

Jason2866, 2021-04-15:
Building master (idf 4.4)  does work. Imho issue can be closed

closed 2021-04-15 by me-no-dev