espressif/esp32-arduino-lib-builder #28: Build an example project faild, which just add components directory .

kylongmu, 2021-04-08:
```
:/mnt/d/Develop/Git_Project/esp32-arduino-lib-builder$ git log
commit 1a838f3b0c06ef34c023ecade1f428eacdd696d1 (HEAD -> master, origin/master, origin/HEAD)
Author: me-no-dev <hristo@espressif.com>
Date:   Mon Apr 5 14:41:59 2021 +0300
```

### Step 1:
            ./build.sh 
            success
### Step 2:
            esp-idf/install.sh 
            . esp-idf/export.sh
            success
### Step 3: select an example "esp-idf/examples/cxx/pthread$"
           idf.py build
           success
### Step 4: modify project file "CMakeLists.txt", add components directory

```
# The following lines of boilerplate have to be in your project's CMakeLists
# in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)
list(APPEND EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/../components/)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(cpp_pthread)

```

 idf.py build
faild with these message:

```
In file included from /mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/RainMaker/src/RMakerParam.h:4,
                 from /mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/RainMaker/src/RMakerDevice.h:4,
                 from /mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/RainMaker/src/RMakerNode.h:4,
                 from /mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/RainMaker/src/RMakerNode.cpp:1:
/mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/RainMaker/src/RMakerType.h:4:10: fatal error: esp_rmaker_core.h: No such file or directory
 #include <esp_rmaker_core.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [esp-idf/arduino/CMakeFiles/__idf_arduino.dir/build.make:778: esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/RainMaker/src/RMakerNode.cpp.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/RainMaker/src/RMakerType.cpp:1:
/mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/RainMaker/src/RMakerType.h:4:10: fatal error: esp_rmaker_core.h: No such file or directory
 #include <esp_rmaker_core.h>
          ^~~~~~~~~~~~~~~~~~~
In file included from /mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/RainMaker/src/RMakerParam.h:4,
                 from /mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/RainMaker/src/RMakerDevice.h:4,
                 from /mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/RainMaker/src/RMakerDevice.cpp:1:
/mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/RainMaker/src/RMakerType.h:4:10: fatal error: esp_rmaker_core.h: No such file or directory
 #include <esp_rmaker_core.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
compilation terminated.
make[2]: *** [esp-idf/arduino/CMakeFiles/__idf_arduino.dir/build.make:817: esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/RainMaker/src/RMakerType.cpp.obj] Error 1
make[2]: *** [esp-idf/arduino/CMakeFiles/__idf_arduino.dir/build.make:804: esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/RainMaker/src/RMakerDevice.cpp.obj] Error 1
In file included from /mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/RainMaker/src/RMakerParam.h:4,
                 from /mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/RainMaker/src/RMakerParam.cpp:1:
/mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/RainMaker/src/RMakerType.h:4:10: fatal error: esp_rmaker_core.h: No such file or directory
 #include <esp_rmaker_core.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [esp-idf/arduino/CMakeFiles/__idf_arduino.dir/build.make:791: esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/RainMaker/src/RMakerParam.cpp.obj] Error 1
In file included from /mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/RainMaker/src/RMakerParam.h:4,
                 from /mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/RainMaker/src/RMakerDevice.h:4,
                 from /mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/RainMaker/src/RMakerNode.h:4,
                 from /mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/RainMaker/src/RMaker.h:5,
                 from /mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/RainMaker/src/RMaker.cpp:1:
[... 50 lines cut ...]
                            ^~~~~
/mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/WiFi/src/WiFiGeneric.cpp: At global scope:
/mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/WiFi/src/WiFiGeneric.cpp:188:21: warning: 'const char* auth_mode_str(int)' defined but not used [-Wunused-function]
 static const char * auth_mode_str(int authmode)
                     ^~~~~~~~~~~~~
/mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/WiFi/src/WiFiScan.cpp:45:21: warning: 'const char* cipher_str(int)' defined but not used [-Wunused-function]
 static const char * cipher_str(int cipher)
                     ^~~~~~~~~~
/mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/WiFiClientSecure/src/ssl_client.cpp:23:4: error: #error "Please configure IDF framework to include mbedTLS -> Enable pre-shared-key ciphersuites and activate at least one cipher"
 #  error "Please configure IDF framework to include mbedTLS -> Enable pre-shared-key ciphersuites and activate at least one cipher"
    ^~~~~
/mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/WiFiClientSecure/src/ssl_client.cpp
 In function 'int start_ssl_client(sslclient_context*, const char*, uint32_t, int, const char*, const char*, const char*, const char*, const char*, bool)':
/mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/WiFiClientSecure/src/ssl_client.cpp:164:15: error: 'mbedtls_ssl_conf_psk' was not declared in this scope
         ret = mbedtls_ssl_conf_psk(&ssl_client->ssl_conf, psk, psk_len,
               ^~~~~~~~~~~~~~~~~~~~
/mnt/d/Develop/espressif/Project/pdm_mic_WiFiUDPClient/components/arduino/libraries/WiFiClientSecure/src/ssl_client.cpp:164:15: note: suggested alternative: 'mbedtls_ssl_conf_sni'
         ret = mbedtls_ssl_conf_psk(&ssl_client->ssl_conf, psk, psk_len,
               ^~~~~~~~~~~~~~~~~~~~
               mbedtls_ssl_conf_sni
make[2]: *** [esp-idf/arduino/CMakeFiles/__idf_arduino.dir/build.make:999: esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/WiFiClientSecure/src/ssl_client.cpp.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:5095: esp-idf/arduino/CMakeFiles/__idf_arduino.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
make failed with exit code 2
```

me-no-dev, 2021-04-08:
you are missing the components and proper sdkconfig :)

closed 2021-04-08 by me-no-dev

kylongmu, 2021-04-08:
> 
> 
> you are missing the components and proper sdkconfig :)
How to add the missing components and poper sdkconfig?