espressif/esp32-arduino-lib-builder #45: how to build 1.0.6

alorbach, 2021-09-24:
Can somebody tell me what branches I need to build only 1.0.6?
Basically I need the last official released version from https://github.com/espressif/arduino-esp32 with Bluetooth Classic and mbedtls DTLS support enabled, thats all I want ;)

alorbach, 2021-09-24:
2.0.0 totally breaks my code (pretty large project for esp32) at certain places. FastLED is not working, network PING and other issues I havent discovered yet. 

So would really be great to build 1.0.6 with these utils, I need DTLS (mbedtls) support badly.

Joshy-2010, 2022-05-15:
Did you figure it out? I have the same problem, my old esp32 project only works with arduino-esp32 version 1.0.6