espressif/esp32-arduino-lib-builder #52: Default header receive buffer size CONFIG_HTTPD_MAX_REQ_HDR_LEN=512 too small

craigdetter, 2021-12-04:
The default in build for the sdkconfig configuration parameter is:
CONFIG_HTTPD_MAX_REQ_HDR_LEN=512 (bytes).

On Google Chrome latest version running on MacOS on the latest Catalina OS:
using Wireshark trace - shows the "post" header is 554 bytes long (not incl. CR/LF)

It needs to be set at least 768 bytes, but 1024 would be better.

See details here:
[https://github.com/espressif/arduino-esp32/issues/5969](url)

LenShustek, 2022-01-14:
This problem has been talked about for years in various forums. Will it ever get fixed?

Until this is fixed, a server using arduino-esp32 that implements any web page buttons will fail when the client browser is Chrome or Edge. It works with Firefox and the (obsolete) Internet Explorer, but a that's not a requirement you can impose on users. Please help, or we need to abandon the Arduino IDE!

me-no-dev, 2022-01-18:
> Will it ever get fixed?

Done. Sorry for the delay :)

closed 2022-01-18 by me-no-dev

LenShustek, 2022-01-18:
Wonderful, thanks! When will the change appear in espressif/arduino-esp32 so we can use it?

me-no-dev, 2022-01-18:
Next time that there is something new in ESP-IDF v4.4, the lib-builder will rebuild and create PR to arduino-esp32. When we merge that PR, it will be available through git.

Patsjemoe, 2022-07-05:
Hi,
I was using Firefox, so could get around it.
But now I am using Webview in Android Studio, and have again the problem of headers too long.
Changing the HTTPD_MAX value in sdkconfig does not solve the issue.
Is there a "real" solution in the mean time ?
thanks in advance
best regards
Ludo

DonMorr, 2022-09-23:
> Hi, I was using Firefox, so could get around it. But now I am using Webview in Android Studio, and have again the problem of headers too long. Changing the HTTPD_MAX value in sdkconfig does not solve the issue. Is there a "real" solution in the mean time ? thanks in advance best regards Ludo

Hi @Patsjemoe , did you have any luck sorting this? I'm hitting the exact same issue. Cheers, Donal

commit 5a8939e9 (2022-01-18): Raise CONFIG_HTTPD_MAX_REQ_HDR_LEN to 1024

Fixes: #52