espressif/esp32-arduino-lib-builder #74: ESP32 header fields too long for server to interpret

Patsjemoe, 2022-07-05:
Hi,
This issue has been on GIthub before for a long time, but I still see no solution.
I was using Firefox, so could get around it.
But now I am using Webview in Android Studio, and have again the problem of headers too long.
Changing the HTTPD_MAX value in sdkconfig does not solve the issue.
Is there a "real" solution in the mean time ?
thanks in advance
best regards
Ludo

me-no-dev, 2022-07-06:
Do you have any idea how large are the headers? Did you recompile the libs after changing the value in sdkconfig?

Patsjemoe, 2022-07-06:
Hi,
Thank you for your quick reply.
I recompile the program, so I assume this takes the new file. Do you need
to recompile the libraries separately? If so, how?
Regards
Ludo

me-no-dev, 2022-07-06:
Which version of ESP32-Arduino are you using?

https://docs.espressif.com/projects/arduino-esp32/en/latest/lib_builder.html

Patsjemoe, 2022-07-06:
Hi,
I am using arduino 1.8.19
Regards
Ludo

me-no-dev, 2022-07-06:
> I am using arduino 1.8.19

This is the IDE version. I am asking about the ESP32-Arduino version. Something like 1.0.6, 2.0.3 etc.

> I had the impression all these files were downloaded, so how can I include the sdkconfig with my changes ?

You change [this file here](https://github.com/espressif/esp32-arduino-lib-builder/blob/master/configs/defconfig.common#L42) and then compile and copy the libs

Patsjemoe, 2022-07-06:
Where do I find that?

Patsjemoe, 2022-07-06:
Hi,
I found : tags/2.0.4-0-gc93bf11f
THis is in c:/windows/syswow64/esp32-lib-builder.
Can you confirm which libs to copy ? Is this
esp32-arduino-lib-builder/components/arduino/libraries ?
regards
Ludo

Patsjemoe, 2022-07-07:
Hi,
Is there a document /tutorial on the lib builder ?
Specifically , it is not clear to me on how/what to compile in which
directory, and to get the right files to arduino IDE.
Regards
Ludo

Patsjemoe, 2022-07-07:
> ./build.sh
> ./tools/copy-to-arduino.sh

Hi Me No Dev,
is ./tools/copy-to-arduino.sh copying to the installed arduino IDE ?
regards
Ludo

Patsjemoe, 2022-07-07:
Hi,
when I look into this file copy-to-arduino.sh, I see only the ESP32_ARDUINO
directory, and do not find anything
that copies it to the IDE installation. What am I missing ?
regards
Ludo

me-no-dev, 2022-07-07:
are you looking at the script or the result of running it? It is becoming really hard to help you now.

Patsjemoe, 2022-07-07:
Hi,
I was looking to the script, as it is not clear to me where the result is
copied to.
It is also not clear to me what the result is, and how I can get this to my
arduino IDE ?
regards
Ludo

me-no-dev, 2022-07-07:
I think that this conversation is going out of topic :) You wanted to change a value and test it, but now we are into explaining what each script does. The script will detect your OS and select the proper folder for your arduino-esp32 install (which you should have done from github into your sketches/hardware folder). I hope this is enough for you to actually try what I have been asking...

closed 2022-07-09 by Patsjemoe