espressif/esp32-arduino-lib-builder #77: Unable to complete build on master branch

madmijk, 2022-07-20:
Hello! I am trying to compile Arduino libraries for my ESP32-C3-based board with some menuconfig tweaks, but I am running into various issues. I decided to wipe my installations of ESP-IDF and this library and start over. I am running this on a 64-bit Debian virtual machine in VirtualBox on Windows 10.

I followed the following installation instructions:
1. [ESP-IDF v4.4.1 (latest stable version that satisfies the IDF version requirement of the library builder)](https://docs.espressif.com/projects/esp-idf/en/v4.4.1/esp32c3/get-started/index.html) - though I ran `./install.sh all`
2. [Library builder](https://docs.espressif.com/projects/arduino-esp32/en/latest/lib_builder.html)

I must also add that the above instructions for the library builder are more up-to-date than the readme provided in the root of this Github repository. The latter suggests to install some packages that no longer exist.

But now that I run `./build.sh`, I get this error:

```
-- Building ESP-IDF components for target esp32s3

CMake Error at /home/mike/esp/esp-idf/tools/cmake/build.cmake:455 (message):

  Traceback (most recent call last):

    File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
      return _run_code(code, main_globals, None,

    File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
      exec(code, run_globals)

    File "/home/mike/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/idf_component_manager/prepare_components/__main__.py", line 3, in <module>
      main()

    File "/home/mike/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/idf_component_manager/prepare_components/prepare.py", line 126, in main
      args.func(args)

    File "/home/mike/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/idf_component_manager/prepare_components/prepare.py", line 39, in prepare_dep_dirs
      ComponentManager(args.project_dir).prepare_dep_dirs(

    File "/home/mike/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/idf_component_manager/core.py", line 401, in prepare_dep_dirs
      manifests.append(ManifestManager(component['path'], component['name']).load())

    File "/home/mike/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/idf_component_tools/manifest/manager.py", line 129, in load
      self.check_filename().validate()

    File "/home/mike/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/idf_component_tools/manifest/manager.py", line 68, in validate
      self.manifest_tree, check_required_fields=self.check_required_fields, version=self.version)

    File "/home/mike/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/idf_component_tools/manifest/manager.py", line 92, in manifest_tree
      self._manifest_tree = self.parse_manifest_file()

    File "/home/mike/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/idf_component_tools/manifest/manager.py", line 122, in parse_manifest_file
      return _convert_env_vars_in_yaml_dict(manifest_data)

    File "/home/mike/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/idf_component_tools/manifest/manager.py", line 35, in _convert_env_vars_in_yaml_dict
      d[k] = _convert_env_vars_in_yaml_dict(v)

    File "/home/mike/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/idf_component_tools/manifest/manager.py", line 35, in _convert_env_vars_in_yaml_dict
      d[k] = _convert_env_vars_in_yaml_dict(v)

    File "/home/mike/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/idf_component_tools/manifest/manager.py", line 39, in _convert_env_vars_in_yaml_dict
      d[k] = [_convert_env_vars_in_str(i) for i in v]

    File "/home/mike/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/idf_component_tools/manifest/manager.py", line 39, in <listcomp>
      d[k] = [_convert_env_vars_in_str(i) for i in v]

    File "/home/mike/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/idf_component_tools/manifest/manager.py", line 22, in _convert_env_vars_in_str
      return Template(s).substitute(os.environ)

    File "/usr/lib/python3.9/string.py", line 121, in substitute
      return self.pattern.sub(convert, self.template)

  TypeError: expected string or bytes-like object

Call Stack (most recent call first):
  /home/mike/esp/esp-idf/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
See also "/home/mike/esp/esp32-arduino-lib-builder/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```
Does someone know how to solve this issue?

I am not super familiar with building on Linux-based operating systems, but I am having better luck here than when I tried to compile the libraries directly in Windows.

Many thanks in advance!

madmijk, 2022-07-22:
I just found out there was a fix made to the failing script on the _IDF Component Manager_ repository: [Fix env variable expansion](https://github.com/espressif/idf-component-manager/commit/f8c8d3d72630235d6cf8ab5d8e11f23db207c7ff).
Do I need to wait for another release on the ESP-IDF repository? When I go on the master branch there, the library builder complains that the IDF version is now 5.0.0, which is why I opted for the stable 4.4.1 release.

Jason2866, 2022-07-28:
Arduino lib builder does work as it is. Do not install IDF or Arduino! Just install the requirements the lib builder is needing. Every IDF / Arduino stuff will be installed from the lib builder.

madmijk, 2022-07-28:
I see! I removed my ESP-IDF installation as well as the `.espressif` directory, and now it builds. Thank you!

closed 2022-07-28 by madmijk