espressif/esp32-arduino-lib-builder #78: Option to disable Bluetooth support

benedekt, 2022-07-25:


VojtechBartoska, 2022-07-25:
Moved to Arduino Lib Builder repository.

me-no-dev, 2022-08-23:
@benedekt you can compile with whatever options you want. In case of errors caused by BT being disabled, please let us know. ESP-IDF v4.4 is currently supported

Jason2866, 2022-09-01:
The Arduino lib builder does not support changes in the sdkconfig files via `idf.py menuconfig`. You have to edit it manually in the corresponding file(s).

SinglWolf, 2022-09-01:
> The Arduino lib builder does not support changes in the sdkconfig files via `idf.py menuconfig`.

In this case, the command: `./build.sh -b menuconfig -t esp32` loses all meaning and only misleads.

Jason2866, 2022-09-01:
Yep, the help text is wrong

me-no-dev, 2022-09-13:
That command is useful when testing a menuconfig change or similar. As long as you have ESP-IDF in your ENV, you can run the command and then run `idf.py build` to build the project (and flash/monitor it as well). For exporting libs, you need to edit the configuration files, but for using the repo as project starter or dev, command is useful