espressif/esp32-arduino-lib-builder #79: Menuconfig is not saved

PilnyTomas, 2022-08-10:
Settings in menuconfig are not saved and will not take effect when built.
Steps to reproduce:
`./build.sh -t esp32 -b menuconfig`
Change any config and save.
`./build.sh -t esp32 -b menuconfig`
Observe that the previous setting is on the default value.
Optionally - build and observe that the setting did not take any effect, and is not mentioned in the built sdkconfig.

PilnyTomas, 2022-09-01:
After discussion with @me-no-dev the propper way is to modify `esp32-arduino-lib-builder/configs/defconfig.common`
See https://github.com/espressif/arduino-esp32/pull/7193

closed 2022-09-01 by PilnyTomas

SinglWolf, 2022-09-01:
> After discussion with @me-no-dev the propper way is to modify `esp32-arduino-lib-builder/configs/defconfig.common`

In that case, what's the point of the command: `./build.sh -t esp 32 -b menuconfig`, if my have to manually edit `esp32-arduino-lib-builder/configs/defconfig.common`????

PilnyTomas, 2022-09-01:
> what's the point of the command `./build.sh -t esp32 -b menuconfig`:

Maybe @me-no-dev can answer that.

chegewara, 2022-10-17:
> > After discussion with @me-no-dev the propper way is to modify `esp32-arduino-lib-builder/configs/defconfig.common`
> 
> In that case, what's the point of the command: `./build.sh -t esp32 -b menuconfig`, if my have to manually edit `esp32-arduino-lib-builder/configs/defconfig.common`????

For some params you have default values, which will overwrite the values you set, but not all. The point is to have default values for espressif build make it easier. If you want to change some value which exist in default.xxx you just have to remove it or comment out in default.xxx file.

It is rare case when users edit values in menuconfig, but its common case (CI) when its required to use default values.
Sure, its not optimal, but i think the only option with Smake to do it, but i may be wrong (not cmake expert).

jcwren, 2022-11-13:
Seems like the rather sparse `README.md` should be updated to note this. I just wasted 2 hours trying to figure out why trying to enable the FreeRTOS trace facility via the `CONFIG_FREERTOS_USE_TRACE_FACILITY` in `sdkconfig` was just getting tossed out like a piece of old dead fish, only to find it in a closed issue.

It's very counterintuitive to have to modify files in `esp32-arduino-lib-builder/configs` when ESP-32 developers (and people with a Linux kernel background) are used to using `menuconfig`.

me-no-dev, 2022-11-14:
`menuconfig` and `sdkconfig` relate only to certain configuration/chip at once. Here we build for all supported chips. The `./build.sh -t [chip] menuconfig` in intended to configure a normal project build for particular chip only and not to relate that to building and exporting libraries to Arduino. Imagine it as you using the repo to build your own firmware for particular board that uses Arduino as component.