espressif/esp32-arduino-lib-builder #81: Docker image

perotom, 2022-08-22:
I am trying to build a docker image to be able to recompile libraries with changed `sdkconfig`. My Dockerfile so far:

```
FROM ubuntu:20.04
ENV TZ=Europe/Vienna
RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone
RUN apt-get update
RUN apt-get install -y git wget curl libssl-dev libncurses-dev flex bison gperf python python-pip python-setuptools python-serial python-click python-cryptography python-future python-pyparsing python-pyelftools cmake ninja-build ccache
RUN pip install --upgrade pip
RUN git clone https://github.com/espressif/esp32-arduino-lib-builder
```

Error:
```
Step 1/7 : FROM ubuntu:20.04
 ---> 3bc6e9f30f51
Step 2/7 : ENV TZ=Europe/Vienna
 ---> Using cache
 ---> 9b789ae46cf0
Step 3/7 : RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone
 ---> Using cache
 ---> 29f72e344dd8
Step 4/7 : RUN apt-get update
 ---> Using cache
 ---> 8c3addb28a00
Step 5/7 : RUN apt-get install -y git wget curl libssl-dev libncurses-dev flex bison gperf python python-pip python-setuptools python-serial python-click python-cryptography python-future python-pyparsing python-pyelftools cmake ninja-build ccache
 ---> Running in 6d9539873edc
Reading package lists...
Building dependency tree...
Reading state information...
Package python-pip is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python3-pip

E: Package 'python-pip' has no installation candidate
E: Unable to locate package python-serial
E: Unable to locate package python-click
E: Unable to locate package python-future
The command '/bin/sh -c apt-get install -y git wget curl libssl-dev libncurses-dev flex bison gperf python python-pip python-setuptools python-serial python-click python-cryptography python-future python-pyparsing python-pyelftools cmake ninja-build ccache' returned a non-zero code: 100
```

When I use Python3 it also doesn't work. Is there an example on how to create a Dockerfile?

igrr, 2022-08-22:
Hi @perotom perhaps you could try using `espressif/idf:release-v4.4` image as the base? Please see the [docs](https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32/api-guides/tools/idf-docker-image.html) and the [source](https://github.com/espressif/esp-idf/blob/release/v4.4/tools/docker/Dockerfile) for more information.

perotom, 2022-08-22:
Ah thanks. With the following Dockerfile I get an infinit loop:

```
FROM espressif/idf:release-v4.4
RUN apt-get update
RUN apt-get install -y jq
RUN pip install --upgrade pip
RUN git clone https://github.com/espressif/esp32-arduino-lib-builder
```

perotom, 2022-08-23:
I didn't rename the directory but maybe related to https://github.com/espressif/esp32-arduino-lib-builder/pull/75?