espressif/esp32-arduino-lib-builder #87: Master branch. Compilation failed. The `esp-sr` component for esp32s2 is missing.

SinglWolf, 2022-09-25:
Master branch. The command `./build.sh -t esp32s2` causes an error:
```cmake
[1352/1398] Building C object esp-idf/esp-sr/CMakeFiles/__idf_esp-sr.dir/src/esp_process_sdkconfig.c.obj
FAILED: esp-idf/esp-sr/CMakeFiles/__idf_esp-sr.dir/src/esp_process_sdkconfig.c.obj 
/home/alex/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc  -I/home/alex/esp/esp32-arduino-lib-builder/build/config -I/home/alex/esp/esp32-arduino-lib-builder/components/esp-sr/esp-tts/esp_tts_chinese/include -I/home/alex/esp/esp32-arduino-lib-builder/components/esp-sr/include/esp32 -I/home/alex/esp/esp32-arduino-lib-builder/components/esp-sr/src/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/newlib/platform_include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/freertos/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/freertos/include/esp_additions/freertos -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/freertos/include/esp_additions -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/esp32s2 -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/port/esp32s2/. -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/port/esp32s2/private_include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/heap/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/log/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/lwip/include/apps -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/lwip/include/apps/sntp -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/include/arch -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/soc/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/soc/esp32s2/. -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/soc/esp32s2/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/hal/esp32s2/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/hal/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/hal/platform_port/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/esp_rom/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/esp_rom/include/esp32s2 -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/esp_rom/esp32s2 -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/esp_common/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/esp_system/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/soc -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/public_compat -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/xtensa/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/xtensa/esp32s2/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/driver/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/driver/esp32s2/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/esp_pm/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/esp_ringbuf/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/efuse/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/efuse/esp32s2/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/vfs/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/esp_wifi/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/esp_event/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/esp_netif/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/esp_eth/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/tcpip_adapter/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/esp_phy/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/esp_phy/esp32s2/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/esp_ipc/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/app_trace/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/esp_timer/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/json/cJSON -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/spiffs/include -I/home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/spi_flash/include -mlongcalls  -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Os -freorder-blocks -Wwrite-strings -fstack-protector -fmacro-prefix-map=/home/alex/esp/esp32-arduino-lib-builder=. -fmacro-prefix-map=/home/alex/esp/esp32-arduino-lib-builder/esp-idf=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.4.2-312-g08fa67fe92\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/esp-sr/CMakeFiles/__idf_esp-sr.dir/src/esp_process_sdkconfig.c.obj -MF esp-idf/esp-sr/CMakeFiles/__idf_esp-sr.dir/src/esp_process_sdkconfig.c.obj.d -o esp-idf/esp-sr/CMakeFiles/__idf_esp-sr.dir/src/esp_process_sdkconfig.c.obj -c /home/alex/esp/esp32-arduino-lib-builder/components/esp-sr/src/esp_process_sdkconfig.c
/home/alex/esp/esp32-arduino-lib-builder/components/esp-sr/src/esp_process_sdkconfig.c:9:20: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 static char *TAG = "SR_SYS";
                    ^~~~~~~~
/home/alex/esp/esp32-arduino-lib-builder/components/esp-sr/src/esp_process_sdkconfig.c: In function 'esp_mn_commands_update_from_sdkconfig':
/home/alex/esp/esp32-arduino-lib-builder/components/esp-sr/src/esp_process_sdkconfig.c:884:42: error: 'esp_mn_iface_t' {aka 'const struct <anonymous>'} has no member named 'get_language'; did you mean 'get_samp_rate'?
     if (strcmp(ESP_MN_CHINESE, multinet->get_language(model_data)) == 0) {
                                          ^~~~~~~~~~~~
                                          get_samp_rate
/home/alex/esp/esp32-arduino-lib-builder/components/esp-sr/src/esp_process_sdkconfig.c:886:49: error: 'esp_mn_iface_t' {aka 'const struct <anonymous>'} has no member named 'get_language'; did you mean 'get_samp_rate'?
     } else if (strcmp(ESP_MN_ENGLISH, multinet->get_language(model_data)) == 0) {
                                                 ^~~~~~~~~~~~
                                                 get_samp_rate
[1354/1398] Building C object esp-idf/esp-sr/CMakeFiles/__idf_esp-sr.dir/src/esp_mn_speech_commands.c.obj
/home/alex/esp/esp32-arduino-lib-builder/components/esp-sr/src/esp_mn_speech_commands.c:8:20: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 static char *TAG = "MN_COMMAND";
                    ^~~~~~~~~~~~
[1355/1398] Building C object esp-idf/esp-sr/CMakeFiles/__idf_esp-sr.dir/src/model_path.c.obj
/home/alex/esp/esp32-arduino-lib-builder/components/esp-sr/src/model_path.c:16:20: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 static char *TAG = "MODEL_LOADER";
                    ^~~~~~~~~~~~~~
/home/alex/esp/esp32-arduino-lib-builder/components/esp-sr/src/model_path.c:17:33: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 static char *SRMODE_BASE_PATH = "/srmodel";
                                 ^~~~~~~~~~
[1357/1398] Building CXX object esp-idf/esp-dsp/CMakeFiles/__idf_esp-dsp.dir/modules/kalman/ekf/common/ekf.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

me-no-dev, 2022-09-26:
Could you maybe check in the esp-sr repository for issue?All errors seem to be there.

SinglWolf, 2022-09-26:
> Could you maybe check in the esp-sr repository for issue?All errors seem to be there.

In the repository `https://github.com/espressif/esp-sr/tree/master/include` there is no directory for `esp32s2`.
Only `esp32` and `esp32s3`.

me-no-dev, 2022-09-26:
It is using the `esp32` directory for S2, which is fine. See [here](https://github.com/espressif/esp-sr/blob/master/CMakeLists.txt#L43). Issue must be elsewhere in that repository.

SinglWolf, 2022-09-26:
> It is using the `esp32` directory for S2, which is fine. See [here](https://github.com/espressif/esp-sr/blob/master/CMakeLists.txt#L43). Issue must be elsewhere in that repository.

I'm sorry, of course, but why should a simple user look for and fix a developer's problem???

me-no-dev, 2022-09-26:
I'm not asking you to fix it, but rather to report it there. The problem you are having is not in the lib-builder and it is an external component to this project.

SinglWolf, 2022-09-27:
> The problem you are having is not in the lib-builder and it is an external component to this project. I'm not asking you to fix it, but rather to report it there.

"The sheriff is not interested in the problems of the Indians"?
People buy a car of a popular model. Everything was fine and suddenly the new cars started having engine problems. The owners turn to the car manufacturer with a claim. In response, they receive this: "We do not manufacture engines for this car. Make claims to the manufacturer of these engines."
Funny, but very relevant.
I went the other way.
I just replaced the component code from the [archive](https://github.com/espressif/esp-sr/archive/refs/tags/v1.0.zip) of the release version and the compilation was successful.

me-no-dev, 2022-09-27:
so you expect the sales team to know why your crankshaft failed? Unless it's a known problem, they can only send you to the service to have it checked, but even if it's known, it's still work for the car service, not for the sales team. Good on you for replacing the engine with older model, without letting the service know that there is a problem :)

closed 2022-09-27 by me-no-dev