espressif/esp32-arduino-lib-builder #88: Master branch. The `-s` and `-t` keys do wrong

SinglWolf, 2022-09-27:
Executing the command: `./build.sh -s`. Result:
```bash
/build.sh: line 162: idf.py: command not found
```
Executing the command: `./build.sh -c /home/alex/.platformio/packages/framework-arduinoespressif32/ -t esp32 esp32s2`. As a result, only `SoC esp32` is compiled!
And is deleted in the  directory `/home/alex/.platformio/packages/framework-arduinoespressif32/tools/` everything except `SoC esp32`!
I tried to execute the command as indicated in the [manual](https://github.com/espressif/arduino-esp32/blob/master/docs/source/guides/core_debug.rst) `./build.sh -c /home/alex/.platformio/packages/framework-arduinoespressif32/ -t esp32 esp32-s2`. The result is the same.

me-no-dev, 2022-09-27:
Don't know why this was in the docs but it's now fixed: https://github.com/espressif/arduino-esp32/commit/2075bd492bf3ae2b0b66027feeb928752dee3326

You can use `-t` only for specific target

closed 2022-09-27 by me-no-dev

SinglWolf, 2022-10-01:
> Don't know why this was in the docs but it's now fixed: [espressif/arduino-esp32@2075bd4](https://github.com/espressif/arduino-esp32/commit/2075bd492bf3ae2b0b66027feeb928752dee3326)

Sorry for the stupid question.
If you deleted an erroneous entry in the documentation, then the rest of the problems are solved automatically?
And what about the key `-c` in the command `./build.sh`?
And why in the directory `/home/alex/.platformio/packages/framework-arduinoespressif32/tools/` other SoC (`esp32`, `esp32s3`, `esp32c3`) are removed when executing the command `./build.sh -c /home/alex/.platformio/packages/framework-arduinoespressif32/ -t esp32s2`?
Did you close the issue because I'm boring and annoying you a lot?

SinglWolf, 2022-10-21:
19 days have passed, and my question remains unanswered...

me-no-dev, 2022-10-21:
`-c` is meant to replace files in a copy of this repository, not in PIO package. If you want to update your PIO package, you will need to replace the files manually. `-t` specifies which target to build for, so when you specified only one target, all the others were not built and were not in the replaced structure. The copy script removes the whole `tools/sdk` folder. `-c` should not be used with `-t`, unless that is exactly what you intend.

SinglWolf, 2022-10-21:
Executing the command: ./build.sh -s. Result:

```bash
/build.sh: line 162: idf.py: command not found
```
Not a bug, but a feature?

me-no-dev, 2022-10-21:
`-s` required IDF to be installed and in your PATH

SinglWolf, 2022-10-21:
> `-s` required IDF to be installed and in your PATH

How is that?
The IDF is already installed by the script `build.sh`.