espressif/esp32-arduino-lib-builder #89: `SoC esp32s2` Guru Meditation Error...

SinglWolf, 2022-09-27:
Master branch. `Arluino` component branch `2.0.4`. IDE `VsCode+PlatformIO`.
My project lacks internal RAM. Before configuring `esp32-arduino-lib-builder` to suit my needs, I compiled the default libraries. The result is as follows:
```
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40031d41  PS      : 0x00060833  A0      : 0x8003227b  A1      : 0x3fff7900  
A2      : 0x3f780014  A3      : 0x3f780aa8  A4      : 0xffffffff  A5      : 0xffffffff  
A6      : 0x00000006  A7      : 0x3f780a90  A8      : 0x3f780a80  A9      : 0x0000001f  
A10     : 0x00000000  A11     : 0x0000033f  A12     : 0x00000025  A13     : 0x3f780a88  
A14     : 0xfffffffc  A15     : 0x00000019  SAR     : 0x0000001a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x0000000b  LBEG    : 0x00000025  LEND    : 0x3f780a88  LCOUNT  : 0x40028db8  

Backtrace: 0x40031d3e:0x3fff7900 0x40032278:0x3fff7920 0x4003248e:0x3fff7940 0x40027dc1:0x3fff7960 0x40032855:0x3fff7980 0x40156038:0x3fff79a0 0x4014c143:0x3fff79c0 0x400d9c76:0x3fff79e0 0x400888f6:0x3fff7a60 0x40086e14:0x3fff7a80 0x4008a007:0x3fff7ab0

  #0  0x40031d3e:0x3fff7900 in remove_free_block at /home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tlsf.c:213
      (inlined by) block_remove at /home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tlsf.c:261
      (inlined by) block_merge_next at /home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tlsf.c:367
      (inlined by) tlsf_free at /home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tlsf.c:897
  #1  0x40032278:0x3fff7920 in multi_heap_free_impl at /home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:222
  #2  0x4003248e:0x3fff7940 in multi_heap_free at /home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:260
  #3  0x40027dc1:0x3fff7960 in heap_caps_free at /home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:361
  #4  0x40032855:0x3fff7980 in free at /home/alex/esp/esp32-arduino-lib-builder/esp-idf/components/newlib/heap.c:39
  #5  0x40156038:0x3fff79a0 in ieee80211_ioctl at ??:?
  #6  0x4014c143:0x3fff79c0 in esp_wifi_sta_get_ap_info at ??:?
  #7  0x400d9c76:0x3fff79e0 in WiFiSTAClass::RSSI() at /home/alex/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.cpp:717
  #8  0x400888f6:0x3fff7a60 in update_rssi() at main/network.cpp:138
  #9  0x40086e14:0x3fff7a80 in lvgl_core::setup_screen(unsigned char) at main/lvgl_core.cpp:1120
  #10 0x4008a007:0x3fff7ab0 in Services::task_lcd(void*) at main/services.cpp:462 (discriminator 5)
```
When using `framework-arduinoespressif32` without compiling libraries using `esp32-arduino-lib-builder`, this problem did not exist.

me-no-dev, 2022-09-27:
Not clear what you changed in the configuration. Also we do not help with PIO, we can help with Arduino IDE

SinglWolf, 2022-09-27:
> Not clear what you changed in the configuration.

I have already said that I compiled the libraries by default, that is, I did not change anything in the configuration.

>Also we do not help with PIO, we can help with Arduino IDE

Thanks.

SinglWolf, 2022-09-30:
This is my mistake, or rather, inattention coupled with forgetfulness.
Never use `GPIO_NUM_26` if `esp32-s2` has `external RAM`!
`GPIO_NUM_26` is used by the `external RAM` chip and it is impossible to use `GPIO_NUM_26` in your projects, otherwise it will lead to the most unpredictable consequences.

closed 2022-09-30 by SinglWolf