espressif/esp32-arduino-lib-builder #91: How to change some parameters of the TCP?

vernonet, 2022-10-06:
How to change some parameters of the TCP, for example: "CONFIG_LWIP_TCP_TMR_INTERVAL" or "CONFIG_LWIP_TCP_SND_BUF_DEFAULT"?  In the old version of the build.sh script, I changed them in the file "sdkconfig.esp32" and everything compiled perfectly. Now If I use  this for configuration: "./build.sh -t esp32 -b menuconfig" then the file "sdkconfig" is replaced with the default file after compilation.

Jason2866, 2022-10-13:
Related infos in #79

vernonet, 2022-10-18:
I found, some parameters can be changed in the file - _\esp32-arduino-lib-builder\esp-idf\components\lwip\Kconfig_
```
........
config LWIP_TCP_SND_BUF_DEFAULT
            int "Default send buffer size"
            default 5744  # 4 * default MSS
            range 2440 65535 if !LWIP_WND_SCALE
            range 2440 1024000 if LWIP_WND_SCALE
........
```

closed 2022-10-18 by vernonet