espressif/esp32-arduino-lib-builder #94: platformio-build-esp32.py is incomplete after build finishes

jcwren, 2022-11-13:
Using esp-idf v4.4, when `build.sh` completes, the file `./out/tools/platformio-build-esp32.py` is missing everything prior to the `ASFLAGS` line, and everything after then `"_POSIX_READER_WRITER_LOCKS",`  line in the `CPPDEFINES` section.

I'll be happy to provide any additional information, but I'm not sure what's necessary to diagnose this.

[components_arduino_tools_platformio-build-esp32.py.txt](https://github.com/espressif/esp32-arduino-lib-builder/files/9997564/components_arduino_tools_platformio-build-esp32.py.txt)
[out_tools_platformio-build-esp32.py.txt](https://github.com/espressif/esp32-arduino-lib-builder/files/9997565/out_tools_platformio-build-esp32.py.txt)

me-no-dev, 2022-11-14:
you probably got some errors during build, like bad arguments for `sed`. Installing `gsed` and using that can help. What OS are you on?

jcwren, 2022-11-14:
Building on a Mac, using Homebrew. `gsed` is installed.  It appears that it's choking on something about `awk`. I'd tell you what version it is, but there's no `-v` or `--version` option for it. It's in `/usr/bin`, so I suspect it's the Mac OS version. `gawk` is not installed.

```[1780/1780] Generating idf_libs
IDF_TARGET: esp32s3, MEMCONF: qio_qspi, PWD: /Users/jcw/Projects/esp32-arduino-lib-builder, OUT: /Users/jcw/Projects/esp32-arduino-lib-builder/out/tools/sdk/esp32s3
Renaming 'mbedtls' to 'mbedtls_2': /Users/jcw/Projects/esp32-arduino-lib-builder/build/esp-idf/mbedtls/mbedtls/library/libmbedtls.a
awk: syntax error at source line 1
 context is
   >>> /ASFLAGS=\[/{n++}{print>n"pio_start.txt" <<<
awk: illegal statement at source line 1
awk: syntax error at source line 1
 context is
   >>> /"ARDUINO_ARCH_ESP32"/{n++}{print>n"pio_end.txt" <<<
awk: illegal statement at source line 1
cat: pio_start.txt: No such file or directory
rm: pio_end.txt: No such file or directory
rm: 1pio_start.txt: No such file or directory
rm: pio_start.txt: No such file or directory
cat: 1pio_end.txt: No such file or directory
rm: 1pio_end.txt: No such file or directory
awk: syntax error at source line 1
 context is
   >>> /compiler.cpreprocessor.flags.esp32s3=/{n++}{print>n"platform_start.txt" <<<
awk: illegal statement at source line 1
gsed: can't read 1platform_start.txt: No such file or directory
awk: syntax error at source line 1
 context is
   >>> /compiler.ar.flags.esp32s3=/{n++}{print>n"platform_mid.txt" <<<
awk: illegal statement at source line 1
cat: platform_start.txt: No such file or directory
cat: 1platform_mid.txt: No such file or directory
```

me-no-dev, 2022-11-14:
you need `gawk` also. I guess this needs to be checked and added to the scripts and docs :)

jcwren, 2022-11-14:
OK, installed `gawk`, builds correctly now.  I'd advocate documenting that, and aborting the script on any errors.

closed 2022-12-10 by me-no-dev