espressif/esp32-arduino-lib-builder #96: when build to ./build.sh, appear "./build.sh: line 139: jq: command not found"

hyu7000, 2022-11-19:
I was following the install process. [AA](https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/lib_builder.html)

at last step, some error(?) appear. in tools dir, there is not sdk dir.

stdout is not a tty
./build.sh: line 139: jq: command not found

[image]

how to resolve error?

thank you

hamishcunningham, 2022-11-19:
`sudo apt install jq`
should do the trick if you're on Linux?

hyu7000, 2022-11-20:
@hamishcunningham 
i use Window.
should i build it only Linux or Pi?

hamishcunningham, 2022-11-20:
A command ending ".sh" is for a unix type shell, so you need to find the
equivalent ".bat" or similar, or use Linux or an emulator or vm or....

hyu7000, 2022-11-21:
@hamishcunningham 
thank you, i will try ubuntu

closed 2022-11-22 by hyu7000