espressif/esp32-arduino-lib-builder #97: when build, should use only esp-idf cmd?

hyu7000, 2022-11-22:
hi. i'm tring build to change menuconfig.

i use ubuntu of VM and follow this video.
https://www.youtube.com/watch?v=aPXvIInTi4s

at 37:10. "idf.py menuconfig" input to command prompt, but my terminal of ubuntu appear error.

[image]

---------------------------------------------------------------------------------------------------------

hk@hk-VirtualBox:~/components/arduino_demo$ idf.py menuconfig
Executing action: menuconfig
Running cmake in directory /home/hk/components/arduino_demo/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/hk/.espressif/python_env/idf5.1_py3.8_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=0 /home/hk/components/arduino_demo"...
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/hk/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /home/hk/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /home/hk/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/hk/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++
-- Check for working CXX compiler: /home/hk/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
CMake Error at /home/hk/esp/esp-idf/tools/cmake/component.cmake:224 (message):
  CMake Warning (dev) at build_properties.temp.cmake:8:

    Syntax Warning in cmake code at column 47

  

    Argument not separated from preceding token by whitespace.

  Call Stack (most recent call first):

    /home/hk/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:3 (include)

  This warning is for project developers.  Use -Wno-dev to suppress it.

  

  fatal: not a git repository (or any of the parent directories): .git

  CMake Warning (dev) at
  /home/hk/components/arduino_demo/components/arduino/CMakeLists.txt:12 (if):

    if given arguments:

  

      "5.1.0" "AND" "NOT" ""

  

    An argument named "5.1.0" appears in a conditional statement.  Policy
    CMP0012 is not set: if() recognizes numbers and boolean constants.  Run
    "cmake --help-policy CMP0012" for policy details.  Use the cmake_policy
    command to set the policy and suppress this warning.

  Call Stack (most recent call first):

    /home/hk/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:106 (include)
    /home/hk/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:124 (__component_get_requirements)

  This warning is for project developers.  Use -Wno-dev to suppress it.

  

  CMake Error at
  /home/hk/components/arduino_demo/components/arduino/CMakeLists.txt:19
  (message):

    Arduino-esp32 can be used with ESP-IDF versions between 4.4.0 and 4.4.99,
    but a newer version is detected: 5.1.0.

  Call Stack (most recent call first):

    /home/hk/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:106 (include)
    /home/hk/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:124 (__component_get_requirements)

  

  

Call Stack (most recent call first):
  /home/hk/esp/esp-idf/tools/cmake/build.cmake:572 (__component_get_requirements)
  /home/hk/esp/esp-idf/tools/cmake/project.cmake:447 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/home/hk/components/arduino_demo/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

------------------------------------------------------------------------------------------------------------

[CMakeOutput.log](https://github.com/espressif/esp32-arduino-lib-builder/files/10063069/CMakeOutput.log)

different between video and me is terminal. esp-idf command prompt(video) and ubuntu default terminal(me)

must use only esp-idf cmd?

thank you in advance

hamishcunningham, 2022-11-22:
Looks like you have too recent an IDF:

"Arduino-esp32 can be used with ESP-IDF versions between 4.4.0 and 4.4.99,
but a newer version is detected: 5.1.0"

5 is in prerelease and the core isn't ready for it yet. Try the last stable
release of 4?

Jason2866, 2022-11-22:
i do not use menuconfig at all. Just do a git clone and follow the instructions.
Changes in config has to be done by directly editing the sdkconfig files.

closed 2022-11-24 by hyu7000