espressif/esp32-arduino-lib-builder #102: How do I build with a customized sdkconfig

dlarue, 2023-01-06:
I've run idf.py menuconfig and changed the settings but after running "./build.sh" it's reverted back and built with stock config. 
 
I then tried "./build.sh -t esp32 -b menuconfig qio 40m" but it just terminates after it runs menuconfig and I save/exit the config.
Running "./build.sh -s" after results in sdkconfig getting reset to defaults.

My goal is to enable the RTC clock to use an external crystal within the Arduino IDE v2.0.3 environment.

dlarue, 2023-01-08:
do I have it right, I copy components/arduino/tools/sdk/esp32/sdkconfig to root of the libBuilder, then run idf.py menuconfig and setup my configuration, save and exit. Then I copy the sdkconfig back to components/arduino/tools/sdk/esp32 directory before running ./build.sh -t esp32?

me-no-dev, 2023-01-09:
`build.sh -t [target] -b menuconfig` is meant to prepare the project for testing and building a new option only for particular target and should be followed by `idf.py build`. If you want to change an option for the resulting libs, you need to adjust one of the configs in https://github.com/espressif/esp32-arduino-lib-builder/tree/master/configs Which one, depends on if you want to change something for all supported chips or just for particular one. After adjusting the proper config, you can run `build.sh` to generate all libs

dlarue, 2023-01-09:
I will test if doing `build.sh -t [target] -b menuconfig ` followed by `idf.py build` does the trick. I will also look at what changes to  ./configs/defconfig.esp32 could be made. Thank you.