espressif/esp32-arduino-lib-builder #103: LittleFS should be built with -DLFS_NO_DEBUG -DLFS_NO_WARN -DLFS_NO_ERROR

everslick, 2023-01-11:
Otherwise it will pollute the serial output.

https://github.com/espressif/arduino-esp32/issues/7700

igrr, 2023-01-11:
I would instead suggest adding Kconfig options for LittleFS debug level inhttps://github.com/joltwallet/esp_littlefs/blob/master/Kconfig.

Adding compiler macros for littlefs internals seems a bit wrong at arduino-lib-builder level. This is something that should be handled by the component itself.

everslick, 2023-01-11:
> Sorry for redirecting you to another repository, again!

hehe, no worries. I appreciate the input!

igrr, 2023-01-11:
As an alternative, https://github.com/joltwallet/esp_littlefs/blob/f2a949fbb303b93a0ae1cd47f078c49de96a59a7/src/lfs_config.h#L44 could be modified to call ESP_LOGD instead of printf, then the log level can be controlled same way as all other IDF logs are.

(I think I like this option more than keeping printfs and custom Kconfig options.)

everslick, 2023-01-11:
> I think I like this option more than keeping printfs and custom Kconfig options.

me 2. thanks!

everslick, 2023-01-11:
so we should simply replace:
`#define LFS_DEBUG(...) LFS_DEBUG_(__VA_ARGS__, "")`
with:
`#define LFS_DEBUG(...) ESP_LOGD("LFS", __VA_ARGS__)`
correct?

BrianPugh, 2023-05-03:
will be included in esp_littlefs 1.5.4, which should be released later today.

BrianPugh, 2023-05-03:
implemented in the newly released esp_littlefs v1.5.4.

If everyone is happy with it, we can then close this issue.