espressif/esp32-arduino-lib-builder #105: WiP branch ESP IDF 5.1 issues

Jason2866, 2023-02-15:
@me-no-dev  Knowing that the IDF 5.1 build branch is a non stable WiP one i have two questions / issues.
1) Wrong Config
```
lib-builder/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
[493](https://github.com/Jason2866/esp32-arduino-lib-builder/actions/runs/4184112210/jobs/7249335397#step:5:494)
CMake Error at components/arduino/CMakeLists.txt:207 (message):
[494](https://github.com/Jason2866/esp32-arduino-lib-builder/actions/runs/4184112210/jobs/7249335397#step:5:495)
  esp32-arduino requires CONFIG_FREERTOS_HZ=1000 (currently 100)
[495](https://github.com/Jason2866/esp32-arduino-lib-builder/actions/runs/4184112210/jobs/7249335397#step:5:496)

[496](https://github.com/Jason2866/esp32-arduino-lib-builder/actions/runs/4184112210/jobs/7249335397#step:5:497)

[497](https://github.com/Jason2866/esp32-arduino-lib-builder/actions/runs/4184112210/jobs/7249335397#step:5:498)
-- Configuring incomplete, errors occurred!
```

2) later it fails with

```
-- Adding linker script /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
[560](https://github.com/Jason2866/esp32-arduino-lib-builder/actions/runs/4184112210/jobs/7249335397#step:5:561)
CMake Error at esp-idf/tools/cmake/component.cmake:381 (__component_get_property):
[561](https://github.com/Jason2866/esp32-arduino-lib-builder/actions/runs/4184112210/jobs/7249335397#step:5:562)
  __component_get_property Function invoked with incorrect arguments for
[562](https://github.com/Jason2866/esp32-arduino-lib-builder/actions/runs/4184112210/jobs/7249335397#step:5:563)
  function named: __component_get_property
[563](https://github.com/Jason2866/esp32-arduino-lib-builder/actions/runs/4184112210/jobs/7249335397#step:5:564)
Call Stack (most recent call first):
[564](https://github.com/Jason2866/esp32-arduino-lib-builder/actions/runs/4184112210/jobs/7249335397#step:5:565)
  components/esp-sr/CMakeLists.txt:68 (idf_component_get_property)
```

The failing GH Actions build run https://github.com/Jason2866/esp32-arduino-lib-builder/actions/runs/4184112210/jobs/7249335397

Can you help? Thx!

me-no-dev, 2023-02-15:
1. I will fix it, but this is not really an issue. It is used just to pull any packaged components from the registry.
2. The problem is in `esp-sr/CMakeLists.txt`, which expects esp-dsp to come from the registry, while for us it comes from git directly (for now). I have removed [this line](https://github.com/espressif/esp-sr/blob/master/CMakeLists.txt#L68) and replaced [this line](https://github.com/espressif/esp-sr/blob/master/CMakeLists.txt#L74) with just `esp-dsp`.

You can compile after that.

Jason2866, 2023-02-15:
Thx for heads up and showing a solution!

closed 2023-04-11 by Jason2866