espressif/esp32-arduino-lib-builder #108: Reduce Bluetooth heap size by building custom version

garageeks, 2023-02-22:
Hi to all,
I need to use BT A2DP while sending some data on Wi-Fi. When used alone, A2DP works fine.
But with Wi-Fi enabled I was getting some crashes.
I was on 2.0.4, I upgraded to 2.0.6 but still getting crashes.
Upon a bit of investigation, my code with Wi-Fi only have 230KB free heap, but when BT is enabled drops to 40KB and when playing music it drops further to 21KB, which is not stable anymore.
I read somewhere it is possible to recompile BT library binary blob to discard unnecessary functions for a specific project.
For instance, I don't need BLE.
Can you give me a hint on how to do it?

Many thanks,
Nick

ferbar, 2023-04-15:
Add the following lines to configs/defconfig.esp32

```
CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY=y
CONFIG_BT_HFP_ENABLE=n
CONFIG_BT_BLE_ENABLED=n
CONFIG_BT_MULTI_CONNECTION_ENBALE=n
CONFIG_BLE_MESH=n
```

and rebuild the esp32-arduino libraries. This will save some 300kB flash and 50kB ram

0x0fe, 2023-04-22:
A related tquestion on this specific topic, since arduino library builder seems to only work on linux or OSX, is it possible to simply build the bluetooth library in IDF directly (which works under win10), for example from the A2DP example, and just move the blob to ARduino somehow?

ferbar, 2023-04-23:
@0x0fe You could try to copy the sdkconfig to bt_discovery example, menuconfig, build it, copy the libbt.a to your Arduino framework directory. However I bet that you will get wired crashes if you get a version mismatch or different sdkconfig settings that are clashing. I assume that setting up a linux virtual machine will be faster and less frustrating.

0x0fe, 2023-04-24:
@ferbar Thank you, yes i also had the same thought and decided to compile under ubuntu from WSL, it compiles fine, there seems there is some bug with partitions.csv though.

hientv1999, 2023-06-15:
> 

Where can I find this file configs/defconfig.esp32?

0x0fe, 2023-06-15:
@hientv1999 
here https://github.com/espressif/esp32-arduino-lib-builder/tree/master/configs

hientv1999, 2023-06-15:
@0x0fe Thank you. But is there anyway I can do this when using PlatformIO and not ESPIDF?

0x0fe, 2023-06-15:
mmh, yeah, in fact it is quite more convenient, just make a project and include the following in platformio.ini 
then you will have a project with the arduino HAL and the menuconfig directly accessible.
N.B. i am not going to provide step by step support on this, you'll have to figure out.
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#v6.1.0
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.8
framework = 
	arduino
	espidf
board = esp32dev
board_build.f_flash = 80000000L
board_build.f_cpu = 240000000L
board_build.flash_mode = qio
board_upload.maximum_size = 3145728
board_build.partitions = min_spiffs.csv
board_upload.flash_size = 4MB
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
build_flags = 
	-Wdouble-promotion
	-Wimplicit-fallthrough
	-DCONFIG_ARDUHAL_LOG_COLORS=1
	-DBOARD_HAS_PSRAM
	-DCORE_DEBUG_LEVEL=2
	-DCONFIG_MBEDTLS_PSK_MODES=1
	-DCONFIG_MBEDTLS_KEY_EXCHANGE_PSK=1
	-DCONFIG_FREERTOS_HZ=1000
	-DCONFIG_SPIRAM_SPEED_80M
	-DCONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP
	-DBOARD_HAS_1BIT_SDMMC
build_unflags = 
```

hientv1999, 2023-06-15:
> Hi to all, I need to use BT A2DP while sending some data on Wi-Fi. When used alone, A2DP works fine. But with Wi-Fi enabled I was getting some crashes. I was on 2.0.4, I upgraded to 2.0.6 but still [...]

May I know the stability of the signal as WiFi and BLE share the same antenna? Is it still usable?

0x0fe, 2023-06-16:
there is a time domain multiplexing, so bluetooth and wifi can coexist, the problems are more related to ressources, in particular IRAM, you will find out very quickly that once wifi is started, especially with TLS, and bluetooth is active, most of the IRAM is used (depening on which extensions you use in the bluetooth stack) and it is quite tricky to have these functions coexist with the available ressources in many scnarios.

hientv1999, 2023-06-16:
@0x0fe What is the behaviour of the chip when it runs out of IRAM? I never got into this scenario. Will it crash?