espressif/esp32-arduino-lib-builder #110: master branch fails to compile: raincheck compile error

ricardoquesada, 2023-03-27:
Using ESP-IDF v4.4.

I just did:

`./build.sh` from a Raspberri Pi 4

and it fails with.

```
../components/esp-rainmaker/components/esp_rainmaker/src/console/esp_rmaker_console.c:16:10: fatal error: esp_rmaker_common_console.h: No such file or directory
 #include <esp_rmaker_common_console.h>
```

Similar error when tried to compile it from Linux Arch x86_64

JedWW, 2023-03-27:
Same for me.  Sorry for newb question, how do you remove a component?

JedWW, 2023-03-27:
Also this:
[1528/1762] Building CXX object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/RainMaker/src/RMaker.cpp.obj
FAILED: esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/RainMaker/src/RMaker.cpp.obj
.....
esp32-arduino-lib-builder/components/arduino/libraries/RainMaker/src/RMakerType.h:18:10: fatal error: esp_rmaker_core.h: No such file or directory
 #include <esp_rmaker_core.h>
          ^~~~~~~~~~~~~~~~~~~

me-no-dev, 2023-03-27:
Seems that RainMaker team is doing some changes. Issues should be fixed shortly.

SinglWolf, 2023-04-03:
> Seems that RainMaker team is doing some changes. Issues should be fixed shortly.

The issue was closed https://github.com/espressif/esp-rainmaker/issues/254 but the problem remained.
Opened a new issue again https://github.com/espressif/esp-rainmaker/issues/257

ricardoquesada, 2023-04-08:
in the meantime, is there a way to compile for another branch ?

e.g: does it work with branch esp-idf-5 ? what are the steps ? thanks!

JedWW, 2023-04-08:
For my simple purpose, which was to enable NAPT and forwarding in LWIP, I managed to put together a modified version using IDF 4.4.4 in the sketchbook hardware folder.   I was trying to follow the recipe for the lib builder.  IIRC, quite a few things were missing or wrong, like the compiler and python tools, so I overwrote a standard 2.0.7 build with the SDK from the modified lib builder project.  One major problem was that my usual development setup is Visual Studio/Visual Micro on Windows, but the lib builder must be built on Ubuntu.  Ubuntu, IMHO, is horrible for development, but in any case the lib builder produces an Ubuntu version of arduino-esp32, which differs considerably from the Windows version.  I haven't discovered the "official" way to build arduino-esp32 for Windows.

Proceeding merrily along, I discovered my target hardware uses an M5 STAMP_PICO, which for some reason is not a variant board of the standard ESP32, but has it's own board manager ( if that's the term).  Not wanting to go through the lib building pain again, I put a copy of  liblwip.a file with NAPT for the esp32 into  the **Arduino Sketchbook/libraries/**_libraryName_**/src/esp32** folder and added 

 `precompiled=true` 

into the _**_library.properties_**_ file.  I also needed this in my library source file:

```
 #define IP_NAPT 1
 #define IP_FORWARD 1
 #include "lwip/lwip_napt.h"
```
to work around not having the proper sdkconfig settings. This is for an Arduino library.  I believe Visual Micro allows you to put an archive in a **lib** subfolder of a sketch, but I haven't tried it.

ricardoquesada, 2023-04-12:
it seems that it has been fixed. working for me.

closed 2023-04-12 by ricardoquesada