espressif/esp32-arduino-lib-builder #117: error with partition.csv

0x0fe, 2023-04-24:
So, i compiled under ubuntu with **./build.sh -t esp32 -b menuconfig** then **./build.sh**

then i copied/replaced **tools** and **platform.txt** in **C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7** from **esp32-arduino-lib-builder\out**

But when i try to build in the arduino IDE, i get this error related to partitions.csv

```c
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\xxxx\AppData\Local\Arduino15\packages -hardware C:\Users\xxxx\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\xxxx\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\xxxx\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:JTAGAdapter=default,PSRAM=enabled,PartitionScheme=min_spiffs,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none,EraseFlash=none -vid-pid=10C4_EA60 -ide-version=10819 -build-path C:\Users\xxxx\AppData\Local\Temp\arduino_build_503242 -warnings=none -build-cache C:\Users\xxxx\AppData\Local\Temp\arduino_cache_974461 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.riscv32-esp-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.esptool_py.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1 -prefs=runtime.tools.esptool_py-4.5.1.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1 -prefs=runtime.tools.openocd-esp32.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.11.0-esp32-20221026 -prefs=runtime.tools.openocd-esp32-v0.11.0-esp32-20221026.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.11.0-esp32-20221026 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.mklittlefs.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mkspiffs.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -verbose C:\PROJECT\xxxx\FIRMWARE\BS_FB2_TEST_A2DP\BS_FB2_TEST_A2DP.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\xxxx\AppData\Local\Arduino15\packages -hardware C:\Users\xxxx\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\xxxx\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\xxxx\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:JTAGAdapter=default,PSRAM=enabled,PartitionScheme=min_spiffs,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none,EraseFlash=none -vid-pid=10C4_EA60 -ide-version=10819 -build-path C:\Users\xxxx\AppData\Local\Temp\arduino_build_503242 -warnings=none -build-cache C:\Users\xxxx\AppData\Local\Temp\arduino_cache_974461 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.riscv32-esp-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.esptool_py.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1 -prefs=runtime.tools.esptool_py-4.5.1.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1 -prefs=runtime.tools.openocd-esp32.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.11.0-esp32-20221026 -prefs=runtime.tools.openocd-esp32-v0.11.0-esp32-20221026.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.11.0-esp32-20221026 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.mklittlefs.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mkspiffs.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -verbose C:\PROJECT\xxxx\FIRMWARE\BS_FB2_TEST_A2DP\BS_FB2_TEST_A2DP.ino
Using board 'esp32' from platform in folder: C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7
Using core 'esp32' from platform in folder: C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7
cmd /c if exist "C:\\PROJECT\\xxxx\\FIRMWARE\\BS_FB2_TEST_A2DP\\partitions.csv" COPY /y "C:\\PROJECT\\xxxx\\FIRMWARE\\BS_FB2_TEST_A2DP\\partitions.csv" "C:\\Users\\xxxx\\AppData\\Local\\Temp\\arduino_build_503242\\partitions.csv"
cmd /c if not exist "C:\\Users\\xxxx\\AppData\\Local\\Temp\\arduino_build_503242\\partitions.csv" if exist "C:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.7\\variants\\esp32\\partitions.csv" COPY "C:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.7\\variants\\esp32\\partitions.csv" "C:\\Users\\xxxx\\AppData\\Local\\Temp\\arduino_build_503242\\partitions.csv"
cmd /c if not exist "C:\\Users\\xxxx\\AppData\\Local\\Temp\\arduino_build_503242\\partitions.csv" COPY "C:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.7\\tools\\partitions\\min_spiffs.csv" "C:\\Users\\xxxx\\AppData\\Local\\Temp\\arduino_build_503242\\partitions.csv"
Le chemin d'acces specifie est introuvable.
exit status 1
```

0x0fe, 2023-04-27:
so, this issue was caused by replacing the whole **tools** directory from **out**, instead of only replacing the **sdk** directory inside **tools**.

closed 2023-04-27 by 0x0fe

me-no-dev, 2023-04-28:
there is the `./tools/copy-to-arduino.sh` script that does the replacement of files. BTW we are moving away from having the SDK in the core. Starting with 3.0.0, it is an external package, just like toolchain, esptool, etc.

me-no-dev, 2023-04-28:
There are certain things that do need to be in internal RAM always, so not everything will go to PSRAM. As for the script, yes, you are correct , but you can have a look at it and see what goes where

0x0fe, 2023-04-28:
@me-no-dev 

> yes, you are correct , but you can have a look at it and see what goes where
Yes that is what i did actually. 

> so not everything will go to PSRAM
Right, i was however expecting a larger difference.
Here is my config, do i miss something?

```cpp
CONFIG_BTDM_CTRL_MODE_BTDM=y
CONFIG_BTDM_SCAN_DUPL_CACHE_SIZE=20
CONFIG_BT_BTC_TASK_STACK_SIZE=8192
CONFIG_BT_BTU_TASK_STACK_SIZE=8192
CONFIG_BT_CLASSIC_ENABLED=y
CONFIG_BT_A2DP_ENABLE=y
CONFIG_BT_SPP_ENABLED=y
CONFIG_BT_HFP_ENABLE=y
CONFIG_BT_STACK_NO_LOG=y
CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY=y
CONFIG_ESP32_SPIRAM_SUPPORT=y
CONFIG_SPIRAM_OCCUPY_HSPI_HOST=y
CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST=y
CONFIG_BT_ALLOCATION_FROM_SPIRAM_FIRST=y
CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP=y
CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP=y
CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY=y
CONFIG_ESP32_RTC_CLK_SRC_INT_RC=n
CONFIG_ESP32_RTC_CLK_SRC_EXT_CRYS=y
CONFIG_BLE_MESH=n
CONFIG_BT_BLE_BLUFI_ENABLE=y
CONFIG_ESP32_ULP_COPROC_ENABLED=y
CONFIG_ESP32_XTAL_FREQ_AUTO=y
# CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU1 is not set
CONFIG_FREERTOS_FPU_IN_ISR=y
# CONFIG_USE_WAKENET is not set
# CONFIG_USE_MULTINET is not set
CONFIG_TWAI_ERRATA_FIX_BUS_OFF_REC=y
CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST=y
CONFIG_TWAI_ERRATA_FIX_RX_FRAME_INVALID=y
CONFIG_TWAI_ERRATA_FIX_RX_FIFO_CORRUPT=y
CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK=y
```

me-no-dev, 2023-05-02:
looks like all is there. This suggest that while running, most things that could be allocated in PSRAM will be so, but how much and what is relocated there, I could not say.

0x0fe, 2023-05-02:
i hope not, having both the BT and wifi stack allocatiing on SPIRAM in priority i would expect to have more than 90KB free heap on the IRAM, that is not a lot. I will open a tickert in the IDF to get this confirmed.