espressif/esp32-arduino-lib-builder #120: Error when running ./build.sh: AttributeError: 'HTTPResponse' object has no attribute 'strict'

Gspohu, 2023-05-11:
Hello !

I encountered an error when running ./build.sh as suggested in the readme. This is the error message :
```
CMake Error at esp-idf/tools/cmake/build.cmake:463 (message):
  Traceback (most recent call last):

    File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
      return _run_code(code, main_globals, None,
    File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
      exec(code, run_globals)
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_manager/prepare_components/__main__.py", line 6, in <module>
      main()
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_manager/prepare_components/prepare.py", line 124, in main
      args.func(args)
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_manager/prepare_components/prepare.py", line 27, in prepare_dep_dirs
      ComponentManager(
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_manager/core.py", line 62, in wrapper
      return func(self, *args, **kwargs)
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_manager/core.py", line 438, in prepare_dep_dirs
      downloaded_component_paths, downloaded_component_version_dict = download_project_dependencies(
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_manager/dependencies.py", line 111, in download_project_dependencies
      solution = solver.solve()
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_manager/version_solver/version_solver.py", line 34, in solve
      self.solve_manifest(manifest)
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_manager/version_solver/version_solver.py", line 53, in solve_manifest
      self.solve_component(requirement)
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_manager/version_solver/version_solver.py", line 56, in solve_component
      cmp_with_versions = requirement.source.versions(
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_tools/sources/web_service.py", line 143, in versions
      cmp_with_versions = self.api_client.versions(component_name=name, spec=spec)
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_tools/api_client.py", line 285, in wrapper
      return f(self, request=request, *args, **kwargs)
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_tools/api_client.py", line 305, in versions
      body = request(
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_tools/api_client.py", line 282, in request
      return self._base_request(
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_tools/api_client.py", line 213, in _base_request
      response = session.request(
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/requests/sessions.py", line 587, in request
      resp = self.send(prep, **send_kwargs)
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/requests/sessions.py", line 745, in send
      r.content
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/requests/models.py", line 899, in content
      self._content = b"".join(self.iter_content(CONTENT_CHUNK_SIZE)) or b""
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/requests/models.py", line 816, in generate
      yield from self.raw.stream(chunk_size, decode_content=True)
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/urllib3/response.py", line 935, in stream
      data = self.read(amt=amt, decode_content=decode_content)
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/urllib3/response.py", line 874, in read
      data = self._raw_read(amt)
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/urllib3/response.py", line 809, in _raw_read
      data = self._fp_read(amt) if not fp_closed else b""
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/urllib3/response.py", line 794, in _fp_read
      return self._fp.read(amt) if amt is not None else self._fp.read()
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/cachecontrol/filewrapper.py", line 96, in read
      self._close()
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/cachecontrol/filewrapper.py", line 76, in _close
      self.__callback(result)
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/cachecontrol/controller.py", line 353, in cache_response
      self._cache_set(cache_url, request, response, body, expires_time)
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/cachecontrol/controller.py", line 274, in _cache_set
      self.serializer.dumps(request, response, body),
    File "/home/gsp/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/cachecontrol/serialize.py", line 54, in dumps
      u"strict": response.strict,

  AttributeError: 'HTTPResponse' object has no attribute 'strict'

Call Stack (most recent call first):
  esp-idf/tools/cmake/project.cmake:393 (idf_build_process)
  CMakeLists.txt:9 (project)
``` 

The .espressif directory is created by the build.sh script. It seems like there's a problem with the Python environment or some internal process when building the project. I'm using Python version 3.10, if that might be a factor.

This try was on the master version, but it's the same problem on the ip4_napt_support branch.

I would appreciate any help or guidance you can provide to resolve this issue. Thank you.

Gspohu, 2023-05-11:
Maybe it can be link to this issue : https://github.com/espressif/esp-idf/issues/11340

Gspohu, 2023-05-11:
I find a workaround. I added : 
```
import subprocess
import sys

def install_package():
    subprocess.check_call([sys.executable, "-m", "pip", "install", "-U", "urllib3<2"])

install_package()
```

in `.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_tools/sources/idf.py`

At least, I had access to menuconfig. I try to compile now, I will make an update here when finished.

Jason2866, 2023-05-12:
The Arduino libs are built with branch `release/v4.4`

Gspohu, 2023-05-13:
I compile directly master and activating manually on NAPT support.

But anyway @Jason2866 , I get stuck at the last compile step...
```
[1821/1822] Generating idf_libs
```

I tried to wait 17 hours, but nothing happened.

Jason2866, 2023-05-13:
Do NOT use branch master for compiling the Arduino libs for core 2.0.x
USE branch `release/v4.4` as i already wrote before.

Gspohu, 2023-05-13:
In fact, I already tested :
`release/v4.4`
`master`
`ip4_napt_support`

No error message but stuck at `[1818/1819] Generating idf_libs`, for `release/v4.4` for example.

rtrbt, 2023-07-11:
Did you clone this repo into a directory that is not called "esp32-arduino-lib-builder"? If so, the shell script generating the idf_libs is stuck in an infinite loop. Here's the fix: https://github.com/espressif/esp32-arduino-lib-builder/pull/75