espressif/esp32-arduino-lib-builder #124: Build flags across files

HamzaHajeir, 2023-05-30:
I'm trying to build lwip library providing compile flags of `-DLWIP_ALTCP` and `-DLWIP_ALTCP_TLS` to enable TLS over [ALTCP](https://www.nongnu.org/lwip/2_1_x/group__altcp__api.html).

Unfortunately, [esp-lwip doesn't make it possible](https://github.com/espressif/esp-lwip/blob/2.1.2-esp/src/include/lwip/opt.h#L1525) through `CONFIG_LWIP_XXX` configuration, therefore I can't just tweak [defconfig.common](https://github.com/espressif/esp32-arduino-lib-builder/blob/master/configs/defconfig.common) to achieve it.

I've came into conclusion that I need to provide them as compile flags to the builder, how can I achieve it?

With thanks,
Hamza

me-no-dev, 2023-05-30:
Arduino adds defines like this: https://github.com/espressif/arduino-esp32/blob/release/v2.x/CMakeLists.txt#L224-L230C13
But I am not sure if it will propagate across all components, so I suggest you have a look at the build system info: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html

HamzaHajeir, 2023-05-30:
Navigating forward, Got some guidance how to override idf libraries; by copying the library out from idf components directory to the project's components directory.

Will inform with progress.

HamzaHajeir, 2023-05-30:
I've successfully built the libraries with working code, I had to deal with lwip CMakeLists.txt to include altcp sources.

I guess the missing altcp sources to CMakeLists was the source behind previous unsuccessful builds.

me-no-dev, 2023-05-31:
Great!

HamzaHajeir, 2023-06-04:
Just to keep this information until official support of ALTCP:

Enhanced the code with more stabilized ALTCP TCP mode, but with an increased Lwip stack size needed for the enhanced code.

Normal (unenhanced code), which queues a function on main loop thread on data receive for application processing, was working good, but changing to the better code (continue processing within lwip thread just ended up crashing with this decoded log:
```log
13:04:09.160 > Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 

13:04:09.166 > Debug exception reason: Stack canary watchpoint triggered (tiT) 

13:04:09.172 > Core  0 register dump:

13:04:09.174 > PC      : 0x4008db4d  PS      : 0x00060036  A0      : 0x4008bfb7  A1      : 0x3ffb2340  

13:04:09.182 > A2      : 0x3ffc5644  A3      : 0x3ffc5658  A4      : 0x3ffc5658  A5      : 0x00000001  

13:04:09.190 > A6      : 0x00000000  A7      : 0x00000018  A8      : 0x3ffc564c  A9      : 0x3ffb2320  

13:04:09.198 > A10     : 0x3ffb9454  A11     : 0x3ffb9454  A12     : 0x00000014  A13     : 0x00000000  

13:04:09.206 > A14     : 0x3ffb7a28  A15     : 0x80000001  SAR     : 0x00000018  EXCCAUSE: 0x00000001  

13:04:09.213 > EXCVADDR: 0x00000000  LBEG    : 0x400894cc  LEND    : 0x400894d7  LCOUNT  : 0x00000000  

13:04:09.221 > 

13:04:09.221 > 

13:04:09.222 > Backtrace: 0x4008db4a:0x3ffb2340 0x4008bfb4:0x3ffb2370 0x4008bf64:0xa5a5a5a5 |<-CORRUPTED

13:04:09.311 > 
13:04:09.311 >   #0  0x4008db4a:0x3ffb2340 in taskSelectHighestPriorityTaskSMP at /home/hamza/arduino/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3436

13:04:09.311 >       (inlined by) vTaskSwitchContext at /home/hamza/arduino/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3519
13:04:09.311 >   #1  0x4008bfb4:0x3ffb2370 in _frxt_dispatch at /home/hamza/arduino/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/portasm.S:436
13:04:09.311 >   #2  0x4008bf64:0xa5a5a5a5 in _frxt_int_exit at /home/hamza/arduino/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/portasm.S:231
```

Which after further research, found it relates to limited lwip stack size. Thus rebuilt libraries with increased Lwip stack size of 3072 (up from 2560) and it successfully worked.

I'll inform with further notes in my progress upgrading to TLS.