espressif/esp32-arduino-lib-builder #126: error: implicit declaration of function 'esp_rmaker_cmd_prepare_empty_response'

HamzaHajeir, 2023-06-18:
Hi there
I'm on a branch out of release/v4.4 witch in sync with it, and which adds a customized lwip as a component, lately trying to build results in a compile error.

Investigation shows the related header is properly included. Explicitly declaring the function within the file in response `esp_rmaker_cmd_resp_manager.c` solves the error.

Is it common across builders? 
Any idea how to fix it?

Regards,
Hamza 

```log
/home/hamza/arduino/esp32-arduino-lib-builder/components/esp-rainmaker/components/esp_rainmaker/src/core/esp_rmaker_cmd_resp_manager.c: In function 'esp_rmaker_cmd_resp_check_pending':
/home/hamza/arduino/esp32-arduino-lib-builder/components/esp-rainmaker/components/esp_rainmaker/src/core/esp_rmaker_cmd_resp_manager.c:110:9: error: implicit declaration of function 'esp_rmaker_cmd_prepare_empty_response'; did you mean 'esp_rmaker_cmd_resp_parse_response'? [-Werror=implicit-function-declaration]
```

me-no-dev, 2023-06-21:
This is probably a temporary issue within RainMaker. On the 5.1 branch we moved some components around, but that was just so we can continue our other work. You can remove rainmaker component if you do not need it and it will work.

HamzaHajeir, 2023-06-21:
Ok great, I'm not alone 😆.

I've temporarily copied the headers so code can find them.

Thanks,

HamzaHajeir, 2024-05-30:
I don't think this is replicatable for newer version. Closing.

closed 2024-05-30 by HamzaHajeir