espressif/esp32-arduino-lib-builder #127: Linking error "esp_phy/lib/esp32c3/libbtbb.a: file not recognized: file format not recognized"

HamzaHajeir, 2023-06-30:
Hi there,

I was building OK by the tool, targeting esp32 only, however I've encountered a linking error once got to build all targets, seems to be esp32c3 is the target with this error.
Built with a modified branch out of `release/v4.4` one.

Log of linking error:
```log
[1741/1742] Linking CXX executable arduino-lib-builder.elf
FAILED: arduino-lib-builder.elf 
: && /home/hamza/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-g++ -march=rv32imc -nostartfiles -march=rv32imc --specs=nosys.specs CMakeFiles/arduino-lib-builder.elf.dir/project_elf_src_esp32c3.c.obj -o arduino-lib-builder.elf  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/console/libconsole.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/riscv/libriscv.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/asio/libasio.a  esp-idf/bt/libbt.a  esp-idf/cbor/libcbor.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/coap/libcoap.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_https_server/libesp_https_server.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/mdns/libmdns.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/json/libjson.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/spiffs/libspiffs.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/main/libmain.a  esp-idf/rmaker_common/librmaker_common.a  esp-idf/esp_diagnostics/libesp_diagnostics.a  esp-idf/esp_diag_data_store/libesp_diag_data_store.a  esp-idf/esp_insights/libesp_insights.a  esp-idf/rtc_store/librtc_store.a  esp-idf/json_parser/libjson_parser.a  esp-idf/json_generator/libjson_generator.a  esp-idf/esp_schedule/libesp_schedule.a  esp-idf/espressif__esp_secure_cert_mgr/libespressif__esp_secure_cert_mgr.a  esp-idf/esp_rainmaker/libesp_rainmaker.a  esp-idf/gpio_button/libgpio_button.a  esp-idf/qrcode/libqrcode.a  esp-idf/ws2812_led/libws2812_led.a  esp-idf/arduino/libarduino.a  esp-idf/esp32-camera/libesp32-camera.a  esp-idf/esp_littlefs/libesp_littlefs.a  esp-idf/espressif__esp-dsp/libespressif__esp-dsp.a  esp-idf/fb_gfx/libfb_gfx.a  -Wl,--cref  -Wl,--Map="/home/hamza/arduino/esp32-arduino-lib-builder/build/arduino-lib-builder.map"  -Wl,--gc-sections  -fno-rtti  -fno-lto  esp-idf/asio/libasio.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/coap/libcoap.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/libsodium/liblibsodium.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/fatfs/libfatfs.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/openssl/libopenssl.a  esp-idf/spiffs/libspiffs.a  esp-idf/esp_insights/libesp_insights.a  esp-idf/cbor/libcbor.a  esp-idf/esp_diagnostics/libesp_diagnostics.a  -Wl,--wrap=esp_log_write  -Wl,--wrap=esp_log_writev  -Wl,--wrap=log_printf  esp-idf/esp_diag_data_store/libesp_diag_data_store.a  esp-idf/esp_rainmaker/libesp_rainmaker.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/esp_https_server/libesp_https_server.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/protocomm/libprotocomm.a  esp-idf/bt/libbt.a  -L/home/hamza/arduino/esp32-arduino-lib-builder/esp-idf/components/bt/controller/lib_esp32c3_family/esp32c3  -lbtdm_app  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/mdns/libmdns.a  esp-idf/json/libjson.a  esp-idf/json_parser/libjson_parser.a  esp-idf/json_generator/libjson_generator.a  esp-idf/esp_schedule/libesp_schedule.a  esp-idf/espressif__esp_secure_cert_mgr/libespressif__esp_secure_cert_mgr.a  esp-idf/qrcode/libqrcode.a  esp-idf/rmaker_common/librmaker_common.a  esp-idf/mqtt/libmqtt.a  -u _Z5setupv -u _Z4loopv  -L /home/hamza/arduino/esp32-arduino-lib-builder/components/esp-dl/lib/esp32c3  -lcat_face_detect  -lhuman_face_detect  -lcolor_detect  -lmfn  -ldl  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/console/libconsole.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/riscv/libriscv.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/mbedtls/mbedtls/library/libmbedtls.a  esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a  esp-idf/mbedtls/mbedtls/library/libmbedx509.a  ../esp-idf/components/esp_wifi/lib/esp32c3/libcoexist.a  ../esp-idf/components/esp_wifi/lib/esp32c3/libcore.a  ../esp-idf/components/esp_wifi/lib/esp32c3/libespnow.a  ../esp-idf/components/esp_wifi/lib/esp32c3/libmesh.a  ../esp-idf/components/esp_wifi/lib/esp32c3/libnet80211.a  ../esp-idf/components/esp_wifi/lib/esp32c3/libpp.a  ../esp-idf/components/esp_wifi/lib/esp32c3/libsmartconfig.a  ../esp-idf/components/esp_wifi/lib/esp32c3/libwapi.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/console/libconsole.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/li [... cut ...] b_include_pthread_impl  -u newlib_include_assert_impl  -u __cxa_guard_dummy  -lstdc++  esp-idf/pthread/libpthread.a  -lgcc  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  -lgcov  esp-idf/app_trace/libapp_trace.a  -lgcov  -lc && :
/home/hamza/arduino/esp32-arduino-lib-builder/esp-idf/components/esp_phy/lib/esp32c3/libbtbb.a: file not recognized: file format not recognized
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

me-no-dev, 2023-07-18:
temporary issue? do you still have this issue?

HamzaHajeir, 2023-07-21:
Yes,

The command line is:
```
./build.sh -I release/v4.4 -A idf-release/v4.4
```

HamzaHajeir, 2023-07-22:
Same for the command: 
> ./build.sh -I release/v4.4 -A release/v2.x

> FAILED: arduino-lib-builder.elf
> : && /home/hamza/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-g++ -march=rv32imc -nostartfiles -march=rv32imc [...]

closed 2024-05-30 by HamzaHajeir