espressif/esp32-arduino-lib-builder #134: Fails to build on Ubuntu 22 with dependency issue

blind-oracle, 2023-07-31:
Ubuntu 22.

Does not build with the following error:
```
CMake Error at esp-idf/tools/cmake/build.cmake:463 (message):
  ERROR: Because project depends on idf (>=5.0.0,<5.1.0) which doesn't match
  any versions, version solving failed.
```

The same happens on MacOS

rtrbt, 2023-08-02:
Same here. Are you building the release/v4.4 branch? [esp-dl](https://github.com/espressif/esp-dl) released a new version (2.0) that requires ESP-IDF 5.0. Running
```
git switch release/v1.1
```
in components/esp-dl fixed this for me.

blind-oracle, 2023-08-02:
Thanks! But now it fails with 
```
-- Build files have been written to: /usr/src/esp32-arduino-lib-builder/build
make: *** No rule to make target 'idf_libs'.  Stop.
command "idf_libs" is not known to idf.py and is not a Unix Makefiles target
```

EgorBelkov, 2023-08-04:
Debian 12 - same problem. As I understand it, this is due to the fact that git downloads the "master" version, which points to release 4.4, not 5.1