espressif/esp32-arduino-lib-builder #137: Unknown targets: esp32c6

HamzaHajeir, 2023-09-20:
Hi there,

I'm trying to build with the command:
```
./build.sh -I release/v5.1 -A idf-release/v5.1
```

but faced the error: 

> CMake Error at esp-idf/tools/cmake/build.cmake:463 (message):
>   ERROR: 2 problems were found in the manifest file
>   /home/hamza/arduino/esp32-arduino-lib-builder/components/arduino/idf_component.yml:
> 
> 
>   Invalid manifest format
> 
>   Unknown targets: esp32c6
> 
>   
> 
>   SUGGESTION: This component may be using a newer version of the component
>   manager.
> 
>   You can try to update the component manager by running:
> 
>       /home/hamza/.espressif/python_env/idf4.4_py3.10_env/bin/python -m pip install --upgrade idf-component-manager
> 
>   
> 
> Call Stack (most recent call first):
>   esp-idf/tools/cmake/project.cmake:393 (idf_build_process)
>   CMakeLists.txt:12 (project)

I've upgraded idf-component-manager and the same error occured again.

What is the problem? and potentially the fix?

Best regards,
Hamza Hajeir

Jason2866, 2023-09-29:
This is general wrong
```
  /home/hamza/.espressif/python_env/idf4.4_py3.10_env/bin/python -m pip install --upgrade idf-component-manager
```
Since you have an idf4.4 python env active. Can't work.

HamzaHajeir, 2023-09-30:
Thanks for pointing this out @Jason2866, I just remembered some about the environment, I'll check how to solve it, as I guess there should've been an installation of a complete idf5.0 environment.

HamzaHajeir, 2023-09-30:
I think I've bypassed the issue. Thanks @Jason2866

closed 2023-09-30 by HamzaHajeir