espressif/esp32-arduino-lib-builder #138: qrcode.h not found

HamzaHajeir, 2023-09-30:
Hi 

Tried to build `idf-release/v5.1` and got the error indicating a missing `qrcode.h`, however I've just [added it](https://components.espressif.com/components/espressif/qrcode) as a dependency through `idf.py` script: 
```
python3 [IDF-PATH]/idf.py add-dependency "espressif/qrcode^0.1.0"
```

But got another error indicating it's not mentioned within arduino requirements:

> Compilation failed because RMakerQR.h (in "arduino" component) includes qrcode.h, provided by espressif__qrcode component(s).
> However, espressif__qrcode component(s) is not in the requirements list of "arduino".
> To fix this, add espressif__qrcode to REQUIRES list of idf_component_register call in /home/hamza/arduino/esp32-arduino-lib-builder/components/arduino/CMakeLists.txt.

Added it to the line:
```
set(requires spi_flash mbedtls mdns wifi_provisioning wpa_supplicant esp_adc esp_eth http_parser espressif__qrcode)
```

Got the following error:

> /home/hamza/arduino/esp32-arduino-lib-builder/components/arduino/libraries/RainMaker/src/RMakerQR.cpp: In function 'void printQR(const char*, const char*, const char*)':
> /home/hamza/arduino/esp32-arduino-lib-builder/components/arduino/libraries/RainMaker/src/RMakerQR.cpp:16:5: error: 'qrcode_display' was not declared in this scope
>    16 |     qrcode_display(payload);
>       |     ^~~~~~~~~~~~~~

Have I added a wrong rcode library? or what is the problem?

With thanks,
Hamza Hajeir

me-no-dev, 2023-10-02:
pull the latest changes in the branch and start again (deleting `components/rainmaker` folder first). It will set the RainMaker to particular commit that will not cause the error you see

closed 2023-10-02 by me-no-dev

HamzaHajeir, 2023-10-02:
Thanks @me-no-dev, I've followed the provided steps and progressed, but received another error repeatedly:

> -- Building ESP-IDF components for target esp32h2
> Processing 7 dependencies:
> [1/7] chmorgan/esp-libhelix-mp3 (1.0.3)
> [2/7] espressif/esp-dsp (1.4.7)
> [3/7] espressif/esp-nn (1.0.1)
> [4/7] espressif/esp_secure_cert_mgr (2.3.1)
> [5/7] espressif/mdns (1.2.1)
> [6/7] espressif/qrcode (0.1.0)
> [7/7] idf (5.1.1)
> CMake Error at esp-idf/tools/cmake/component.cmake:250 (message):
>   ERROR: Cannot process component requirements.  Multiple candidates to
>   satisfy project requirements:
> 
>     requirement: "qrcode" candidates: "qrcode, espressif__qrcode"
> 
> Call Stack (most recent call first):
>   esp-idf/tools/cmake/build.cmake:574 (__component_get_requirements)
>   esp-idf/tools/cmake/project.cmake:547 (idf_build_process)
>   CMakeLists.txt:12 (project)
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/hamza/arduino/esp32-arduino-lib-builder/build/CMakeFiles/CMakeOutput.log".
> cmake failed with exit code 1, output of the command is in the /home/hamza/arduino/esp32-arduino-lib-builder/build/log/idf_py_stderr_output_43167 and /home/hamza/arduino/esp32-arduino-lib-builder/build/log/idf_py_stdout_output_43167

lancas17, 2023-10-02:
I was able to get it to build by using an older commit from rainmaker (from the last date I used the lib builder), editing update-components.sh:

``` bash
#
# CLONE/UPDATE ESP-RAINMAKER
#
echo "Updating ESP-RainMaker..."
if [ ! -d "$AR_COMPS/esp-rainmaker" ]; then
    git clone $RMAKER_REPO_URL "$AR_COMPS/esp-rainmaker" && \
	git -C "$AR_COMPS/esp-rainmaker" checkout 0414a8530ec1ac8714269302503c71c238b68836 # added this line
    git -C "$AR_COMPS/esp-rainmaker" submodule update --init --recursive
else
	git -C "$AR_COMPS/esp-rainmaker" fetch && \
	git -C "$AR_COMPS/esp-rainmaker" pull --ff-only && \
    git -C "$AR_COMPS/esp-rainmaker" submodule update --init --recursive
fi
if [ $? -ne 0 ]; then exit 1; fi
```

HamzaHajeir, 2023-10-03:
> I was able to get it to build by using an older commit from rainmaker (from the last date I used the lib builder), editing update-components.sh:
> 
> ```shell
> #
> # CLONE/UPDATE ESP-RAINMAKER
> #
> echo "Updating ESP-RainMaker..."
> if [ ! -d "$AR_COMPS/esp-rainmaker" ]; then
>     git clone $RMAKER_REPO_URL "$AR_COMPS/esp-rainmaker" && \
> 	git -C "$AR_COMPS/esp-rainmaker" checkout 0414a8530ec1ac8714269302503c71c238b68836 # added this line
>     git -C "$AR_COMPS/esp-rainmaker" submodule update --init --recursive
> else
> 	git -C "$AR_COMPS/esp-rainmaker" fetch && \
> 	git -C "$AR_COMPS/esp-rainmaker" pull --ff-only && \
>     git -C "$AR_COMPS/esp-rainmaker" submodule update --init --recursive
> fi
> if [ $? -ne 0 ]; then exit 1; fi
> ```

Unfortunately, this didn't help. Still getting the same error descibed in my last comment.

HamzaHajeir, 2023-10-04:
I had to revert the following command, found it in a file and just deleted it, now I've bypassed this error.

> python3 [IDF-PATH]/idf.py add-dependency "espressif/qrcode^0.1.0"

me-no-dev, 2023-10-04:
you found the command that you added yourself :) yeah. You do not need it

abstractBoyd, 2024-10-30:
> I was able to get it to build by using an older commit from rainmaker (from the last date I used the lib builder), editing update-components.sh:
>
> ```shell
> #
> # CLONE/UPDATE ESP-RAINMAKER
> #
> echo [...]

Oh man, that is EXACTLY what I needed, I'm tired of builds working one day and breaking the next, so I'm just adding in specific commit or release versions for all the dependent libraries.  I'm used to java development and gradle/maven dependency management.  This will have to do for now :D

Thanks!