espressif/esp32-arduino-lib-builder #142: zigbee

lbernstone, 2023-10-21:
Please add [zigbee](https://components.espressif.com/components/espressif/esp-zboss-lib) and [zboss](https://components.espressif.com/components/espressif/esp-zboss-lib) as components.  These have variant-specific pre-built library archives, so will be very difficult to integrate otherwise.

VojtechBartoska, 2023-11-01:
Hi, @P-R-O-C-H-Y is now working on that.

Here is related issue in the core, https://github.com/espressif/arduino-esp32/issues/8807

me-no-dev, 2023-11-01:
> These have variant-specific pre-built library archives, so will be very difficult to integrate otherwise.

also complicated to do here :D

VojtechBartoska, 2023-12-07:
@P-R-O-C-H-Y can we close this in favor of https://github.com/espressif/esp32-arduino-lib-builder/pull/146 and https://github.com/espressif/arduino-esp32/pull/8844?

lbernstone, 2023-12-21:
Good job, @P-R-O-C-H-Y

closed 2023-12-21 by lbernstone