espressif/esp32-arduino-lib-builder #144: Error with copy-to-arduino.sh

brentru, 2023-10-26:
@me-no-dev  Within `copy-to-arduino.sh`, there is a folder referenced which does not exist on arduino-esp32 `master`:
`rm -rf $ESP32_ARDUINO/tools/esp32-arduino-libs`
via https://github.com/espressif/esp32-arduino-lib-builder/blob/master/tools/copy-to-arduino.sh#L22

When looking at this folder in the BSP itself, there is no such `esp32-arduino-libs` folder within this directory:
https://github.com/espressif/arduino-esp32/tree/master/tools

me-no-dev, 2023-10-27:
that is such because that is a package that you install through `get.py` in your normal Arduino install. `copy-to-arduino.sh` is used to overwrite those files with the ones you built. So if you have your ESP32-Arduino installed in `[sketches]/hardware/espressif/esp32`, then that script will overwrite the idf libs in it (that you installed by running `get.py`) with the ones you built with the lib-builder

closed 2023-10-27 by me-no-dev

brentru, 2023-10-27:
@me-no-dev Thank you for the clarification!