espressif/esp32-arduino-lib-builder #150: Linker error due to zigbee

brentru, 2023-11-16:
@me-no-dev After running the absolute latest `build.sh` from `master` and then copying to Arduino (`./tools/copy-to-arduino.sh`), I get the following error when attempting to build:

```
xtensa-esp32s2-elf-g++: warning: {build.zigbee_mode}: linker input file unused because linking not done
xtensa-esp32s2-elf-g++: error: {build.zigbee_mode}: linker input file not found: No such file or directory
```

I am curious if it was caused by this PR, https://github.com/espressif/esp32-arduino-lib-builder/pull/146, and if I can remove it for the time being until the arduino esp32 BSP has zigbee as a managed component.

me-no-dev, 2023-11-17:
fix is easy: just add the following two lines to `platform.txt`: https://github.com/espressif/arduino-esp32/pull/8881/files#diff-5c378d0844f0422d4d813eabe83f61ffce82014d8b8aa3e93ee35ce2ca14ca7bR89-R90

brentru, 2023-11-17:
Works perfectly, thank you

closed 2023-11-17 by brentru