espressif/esp32-arduino-lib-builder #192: Unable to Build Arduino ESP32 Library

polarix2017, 2024-07-15:
I'm encountering issues building any versions of the Arduino ESP32 library. I've carefully followed the steps in the documentation. Below are the details of my setup and the error I'm encountering.

**Setup**:

Docker version: Docker version 24.0.2, build cb74dfc
OS:   Ubuntu 18.04.6 LTS (freshly installed)
Steps followed: [docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/lib_builder.html?highlight=lib%20builder#docker-image)

**Steps to Reproduce:**
Run the build script using ./build.sh -t esp32s3 -D default -c /arduino-esp32 -A release/v2.x -I release/v4.4 configured by UI interface.
Observe the output and note the warnings and errors related to the "idf" component.

Expected Behavior:
The Arduino ESP32 library should build successfully without errors.

**Actual Behavior:**
The build process fails with the error message indicating that the project depends on idf (>=5.1), which doesn't match any available versions.

**Additional Context:**
The error suggests that the required version of the idf component is not available.
Attached are the screenshots of the starting process and the error.

**Screenshots**:
[image]
[image]

polarix2017, 2024-07-16:
An other try:
[image]
[image]
[image]

me-no-dev, 2024-07-16:
You can not use the interface to build 2.x. You need to switch to the 4.4 branch of the lib-builder and use `build.sh` directly. The interface is to build the current libs.

cc @lucasssvaz

polarix2017, 2024-07-16:
I followed the instructions from the [ESP32 Arduino library builder documentation](https://docs.espressif.com/projects/arduino-esp32/en/latest/lib_builder.html?highlight=lib%20builder#installing) to use the standalone build.sh.

However, during the pip install process, it enters an infinite loop of backtracking and eventually leads to an error related to tqdm-x.y.z-py2.py3-none-any.whl. The process never completes successfully.

Could someone assist with resolving this issue?

[image]

lucasssvaz, 2024-07-16:
@polarix2017 This is normal for IDF4. It is not stuck, just takes a very long time.