espressif/esp32-arduino-lib-builder #198: How works the configuration of the config files?

cicciocb, 2024-08-04:
Sorry for that, it seems a very simple question but how implement the speed of the flash and the psram at 120MHZ?

I tried to modify the existing defconfig.esp32s3 file but the generation is always the same.

Should  I change the build.json file?

Jason2866, 2024-08-09:
It depends, for which MCU?

cicciocb, 2024-08-09:
> It depends, for which MCU?

Thanks for your answer.
Is for the ESP32-S3

Jason2866, 2024-08-09:
SPIRAM 120Mhz is already implemented and useable. Flash speed of 120Mhz is an experimental feature not implemented in the builder. It needs a board with verified working 120Mhz flash chips. Will end in crash when trying with boards widely available and sold.

To get the needed settings do a simple "Hello World" with native IDF select S3 and enable experimental features. Now you see the new options for 120Mhz. Do your settings you want.
Extract the needed settings and add them in the corresponding configs add an new extra entry in `builds.json`

I did this and tried and ended in crashes (no 120Mhz flash). Deleted the changes in Arduino Lib Builder, so I don't remember exactly what changed.

cicciocb, 2024-08-09:
> SPIRAM 120Mhz is already implemented and useable. Flash speed of 120Mhz is an experimental feature not implemented in the builder. It needs a board with verified working 120Mhz flash chips. Will end in crash when trying with boards widely available and sold.
> 
> To get the needed settings do a simple "Hello World" with native IDF select S3 and enable experimental features. Now you see the new options for 120Mhz. Do your settings you want. Extract the needed settings from the sdkconfig file and add them in the corresponding configs add an new extra entry in `builds.json`
> 
> I did this and tried and ended in crashes (no 120Mhz flash). Deleted the changes in Arduino Lib Builder, so I don't remember exactly what changed.

In fact using the Arduino 3.0.4, even setting the Flash at 120MHz for the flash I cannot see any changes in terms of performances. As the flash and the PSRAM can also work in SDR or DDR, it becomes very complicate to understand how they are working.
Anyway thanks, I'll check before using IDF as you suggested.

Jason2866, 2024-08-09:
> In fact using the Arduino 3.0.4, even setting the Flash at 120MHz I cannot see any changes in terms of performances. As the flash and the PSRAM can also work in SDR or DDR, it becomes very complicate to understand how they are working.
As per my understanding there are no functions available to receive this information.

The speed change is probably only visible when using an oscilloscope 😄

cicciocb, 2024-08-09:
> The speed change is probably only visible when using an oscilloscope 😄

If is only visible with the scope this means that there is no interest at all :-)

Anyway, I have a scope