espressif/esp32-arduino-lib-builder #199: Improving Wi-Fi file transfer

fjmsouza, 2024-08-05:
Hi,
I need to increase the wifi performance, but where the needed registers?
I was looking for edit something as (I'm not sure):

```
CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM
CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM
CONFIG_ESP32_WIFI_TX_BA_WIN
CONFIG_ESP32_WIFI_RX_BA_WIN
CONFIG_ESP32_WIFI_IRAM_OPT
CONFIG_ESP32_WIFI_RX_IRAM_OPT

CONFIG_TCPIP_RECVMBOX_SIZE 
CONFIG_LWIP_IRAM_OPTIMIZATION

CONFIG_TCP_SND_BUF_DEFAULT
CONFIG_TCP_WND_DEFAULT
CONFIG_TCP_RECVMBOX_SIZE
```

But.: 
What really found in config_editor.py?

```
CONFIG_ESP_WIFI_STATIC_RX_BUFFER_NUM
CONFIG_ESP_WIFI_STATIC_TX_BUFFER_NUM
CONFIG_ESP_WIFI_IRAM_OPT
CONFIG_ESP_WIFI_RX_IRAM_OPT
```

and, in the end of "compile static libraries" the result sdkconfig the searched registers reappear...

Please, can you let clear:

- WHERE are the right registers?
- WHERE put the created folder ("tools")?
- and how to optimize the compilation time?

Thank you for your attention.

fjmsouza, 2024-08-05:
inside of  /home/user/.arduino15/packages/esp32/hardware/esp32/2.0.17/tools/sdk/
a lot of "No such file or directory".

me-no-dev, 2024-08-05:
first you need the `release/v4.4` branch of this repository, not master. Then edit `configs/sdkconfig.common` and add the values that you need. After compilation is done, overwrite `/home/user/.arduino15/packages/esp32/hardware/esp32/2.0.17/tools/sdk/` with the contents of `out/tools/sdk/`. You would also need to overwrite `platform.txt` in the core with the one in the output folder.

fjmsouza, 2024-08-05:
Thank you, @me-no-dev , I hope comeback to reveal my progress

fjmsouza, 2024-08-05:
@me-no-dev , I realized that would be `configs/defconfig.common`, ok?

fjmsouza, 2024-08-05:
My challenge:
esp32(as a client) send a file(40MB) to another esp32(as a server), a loop 5KB aftter 5KB at a http.POST(5KB) until finish this 40MB.
But...40MB transfer is taken 7 minutes. A lot o time!!!
How to deacrease this time?

Obs.: I tried to increase this post to 10KB, but the server reset with a "load phroibted".
There are registers in defconfig.common to achieve this?

I'd apreciate your help.

Jason2866, 2024-08-09:
@fjmsouza  I think it is not a low level problem, which needs changes here. Probably a bottleneck "somewhere" in Arduino code.

fjmsouza, 2024-08-12:
Thank you, @Jason2866!
Me too.

Can anyone give an example with another approach?

The main proposal is a server downloading huge files with better performance.

In the meantime, looking at examples like UploadHugeFile.ino