espressif/esp32-arduino-lib-builder #201: Build failure when using docker container

mskvortsov, 2024-08-12:
```
docker run --rm -it -v $PWD/output:/arduino-esp32 \
  espressif/esp32-arduino-lib-builder:latest ./build.sh -t esp32
```
```
CMake Error at esp-idf/tools/cmake/build.cmake:540 (message):
  ERROR: Some components (espressif/esp-dsp) in the "managed_components"
  directory were modified on the disk since the last run of the CMake.
  Content of this directory is managed automatically.
```
[failure.log](https://github.com/user-attachments/files/16580899/failure.log)

me-no-dev, 2024-08-13:
nothing else should be required. CI runs this every 6 hours with success.

@lucasssvaz PTAL

lucasssvaz, 2024-08-13:
Started happening on master as well https://github.com/espressif/esp32-arduino-lib-builder/actions/runs/10369405267/job/28705079316

lucasssvaz, 2024-08-13:
Idk if something changed in IDF

me-no-dev, 2024-08-13:
@lucasssvaz maybe we can try with older version of DSP

lucasssvaz, 2024-08-13:
> @lucasssvaz maybe we can try with older version of DSP

Let me test

lucasssvaz, 2024-08-13:
Yup, does not work even with older versions

Jason2866, 2024-08-16:
issue is solved in actual Arduino Lib Builder

mskvortsov, 2024-08-17:
It builds now, thank you.

closed 2024-08-17 by mskvortsov