espressif/esp32-arduino-lib-builder #213: Cant get config editor to run

MartinMueller2003, 2024-08-29:
Using latest version of the tool as of Aug 31 2024
Executed ./build.sh -t esp32 without any issues
Cannot execute $ ./tools/config_editor/app.py

I am running on a hyper-v Ubuntu 24.04 instance that is up to date and has all tools installed. I have the GUI installed and running. 
I get the same results using terminal or ssh.

What am I missing?

I get the following output:

Traceback (most recent call last):
  File "/home/martin/esp32-arduino-lib-builder/./tools/config_editor/app.py", line 92, in <module>
    class ConfigEditorApp(App):
  File "/usr/local/lib/python3.12/dist-packages/textual/message_pump.py", line 107, in __new__
    class_obj = super().__new__(cls, name, bases, class_dict, **kwargs)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/textual/app.py", line 703, in __init_subclass__
    raise ValueError(
ValueError: SCREENS should contain a Screen type or callable, not an instance (got instance of MainScreen for 'main')

me-no-dev, 2024-08-30:
@lucasssvaz PTAL

lucasssvaz, 2024-08-30:
Yeah, same thing locally. Probably some breaking change in textual. Will fix and freeze the version in a requirements file

lucasssvaz, 2024-08-30:
Done! Will be fixed as soon as the PR is merged 😄

PR #214 (2024-08-30): fix(UI): Fix breaking changes and lock textual version
Fix the breaking changes introduced by `Textual` and lock its version to avoid issues in the future.

Closes #213
closed 2024-08-30 by me-no-dev