espressif/esp32-arduino-lib-builder #215: Package release doesnt contain some chips anymore

maxgerhardt, 2024-08-30:
From https://raw.githubusercontent.com/espressif/arduino-esp32/master/package/package_esp32_index.template.json taking the URL 

```
          "version": "idf-release_v5.1-e026fd1f",
          "systems": [
            {
              "host": "i686-mingw32",
              "url": "https://github.com/espressif/esp32-arduino-lib-builder/releases/download/idf-release_v5.1/esp32-arduino-libs-idf-release_v5.1-e026fd1f.zip",
```
has the content

[image]

ESP32C2 is missing in action? How's that possible?

maxgerhardt, 2024-08-30:
My bad, see README of Arduino-ESP32. Got confused by build errors in CI complaining about the esp32c2 folder not existing.

closed 2024-08-30 by maxgerhardt

orangeagain, 2024-09-26:
so, where is it?

me-no-dev, 2024-09-27:
C2 is not part of our regular builds. If you want support, you need to compile the libs yourself or use Arduino as IDF component.