espressif/esp32-arduino-lib-builder #218: Build fails on esp32-c3 when CONFIG_LWIP_IPV6 is disabled

noisymime, 2024-09-03:
One of the recommendations made on the Minimizing Binary Size page (https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/performance/size.html#lwip-ipv6) is to disable IPv6 if not required. This is done by turning off option `CONFIG_LWIP_IPV6`. 

Doing this for the esp32-c3 target however causes the build to fail with the error:

```
esp32-arduino-lib-builder/components/arduino/cores/esp32/IPAddress.cpp:29:11: error: 'IP6_NO_ZONE' was not declared in this scope
```

This appears to be occurring for at least the `esp32-c3` target, but is not happening for the `esp32` target.

me-no-dev, 2024-09-04:
This issue comes from Arduino. Please post an issue report there, so we can track it and attempt a fix. IPv6 is tightly integrated in Arduino now

noisymime, 2024-09-04:
> This issue comes from Arduino. Please post an issue report there, so we can track it and attempt a fix. IPv6 is tightly integrated in Arduino now

Thanks for the nudge in the right direction. Looks like it's been decided that ipv6 is mandatory with Arduino 3.0.0: https://github.com/espressif/arduino-esp32/issues/9334

Bit of a shame as it's a fairly heavy component on it's own and probably not needed in the majority of cases.

me-no-dev, 2024-09-09:
it was still intended to be guarded with defines, but we have missed them on a few places