espressif/esp32-arduino-lib-builder #233: Docker - How to enter the correct path on my windows system

weoiss1998, 2024-10-23:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Hi,

many thanks for making the Docker container available.
How exactly do I have to specify the command and in the GUI that it copies the compiled files from Docker to my Windows system?

The standard ESP32 repository of the ArduinoIDE is located under: ‘C:\Users\xxx\AppData\Local\Arduino15\packages\esp32’

me-no-dev, 2024-10-23:
@lucasssvaz PTAL

weoiss1998, 2024-10-25:
I give it a try but got a new error while building: 
I'm using the release/v5.1 Branch from the Lib-Builder.
[image]

lucasssvaz, 2024-10-25:
@me-no-dev Do you know what could cause this error ?

weoiss1998, 2024-10-25:
I've downloaded the release/v5.1 branch directly from here 2 hours ago

lucasssvaz, 2024-10-25:
@weoiss1998 Ah, I see one issue. I forgot to update these files to reflect the proper tag that should be used. Please edit the ps1 file and add `:release-v5.1` to the end of last line. It should be as it follows:

`docker run @($DOCKER_ARGS) espressif/esp32-arduino-lib-builder:release-v5.1`

I will open a PR to fix this

weoiss1998, 2024-10-26:
@lucasssvaz it's working now, thank you for your help. 
I have two additional questions, if it's okay for you. 
1)To run the ps1-Script, I have to open it first with an editor because it's not signed. When I allow the script after editing, I can run the script. Can you sign it via Espressif? If not, I would create a PR to add some hints to the docker docs.

2)Do I have to enter something in the Settings Menu? I leave it blank, and it seems that it took the correct branches.
I would add this hint aswell for future users.

lucasssvaz, 2024-10-28:
@weoiss1998 Will add some notes to the documentation to clarify these points. Thank you!

closed 2024-11-07 by lucasssvaz